ControlSystemStudio/phoebus #787: File browser - Enhancement

DorianHas, 2019-08-19:
Hi there !

Is it possible to hide the File browser toolbar (home button, path input and directory button) like the Data Browser (right click => hide Toolbar) ? It was mentioned as "locked mode" in #288, the idea is to prevent users to change the directory.

By the way, Drag and drop is acting weird. It copies files where I dropped them. But it doesn't remove them from the previous location and displays the following error : 

[image]
 

Dorian

kasemir, 2019-08-19:
Well, as you can see in that ticket we looked at two options: Lock the folder, or offer a home button so users can easily go back to some predefined folder.
We decided to go with the home button option: Users who need to look around can now do that, then easily go back to the home folder.

If you don't want your users to browse the file system, you can remove the file browser from your site-specific installation and only offer them with a "start" display that has all the links they ever need.

As for drag/drop: By default, files are copied. When I try that, there's a "+" decorator on the drag/drop icon to indicate that it will indeed copy, not move.
When I hold the shift key, the drag/drop icon changes, and the file is moved instead of copied.

DorianHas, 2019-08-19:
Thank you ! I will look at it

closed 2019-08-19 by DorianHas

kasemir, 2019-08-19:
It's well possible that there's a drag/drop problem. I also don't remember why 'copy' is the default. maybe 'move' would be a better default.
I think one issue was:
You can drag/drop a *.opi or *.bob file from the file browser into the display editor. This will create a 'related display' button. It will NOT delete the file from the file browser, as it would when the file browser defaults to 'move'.
As it is now, I'm not aware of any drag/drop problems, but if you find them, please describe and ideally also fix them.

While you're at it: I am aware of a UI issue. Sometimes there are empty entries at the bottom of the file browser that show random icons. If you see that, try to reproduce and fix.