ControlSystemStudio/phoebus #789: PV Table - Enhancement

DorianHas, 2019-08-19:
Hi there,

In CSS, there was a PV Table update : https://github.com/ControlSystemStudio/cs-studio/issues/1515

Added features was : 
- Table export to Excel file
- New column Timestamp Save which save the timestamp for each snapshot
- Ability to create a configuration list of PVs if a line start with #conf#
- If a configuration list exists in the table the user can  “make a measurement”. Make a measurement is the ability to take a snapshot of the configuration PV list and add the snapshot at the end of the table.
- The user can delete the last measure or all the measures.

Will these features be integrated into Phoebus too ?

kasemir, 2019-08-19:
Yes, I remember, it was in fact your site which added this measurement option.
If you still need that, feel free to also add it to the phoebus version, but just as in the RCP version, please use a preference setting that allows disabling it.

DorianHas, 2019-08-19:
Alright. I'm adding it to the TODO List.

closed 2019-08-19 by DorianHas

kasemir, 2019-08-19:
The 'model' part should be just copy/paste.
The UI part is different, since we're now using JavaFX instead of SWT.
I didn't try to port that since I really don't need that feature, so cannot justify spending my time on it.
You would also have a better way to test it since you likely have existing files that you can open/compare.