ControlSystemStudio/phoebus #791: Alarm logger configuration & query

MaticPogacnik, 2019-08-21:
I am trying to configure alarm logger in Phoebus but have some problems. It doesn't matter what kind of query I make I can't get any output. 
I tried simple query as `pv=*`

This is output when I press search:
`
[user@host721057 phoebus-product]$ 2019-08-21 08:24:28 INFO [org.phoebus.framework.jobs] Job 'searching alarm log messages for : *': Finished - Cancelled
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 15] start execution
2019-08-21 08:24:28 FINE [org.apache.http.client.protocol.RequestAddCookies] CookieSpec selected: default
2019-08-21 08:24:28 FINE [org.apache.http.client.protocol.RequestAuthCache] Re-using cached 'basic' auth scheme for http://10.72.10.83:9200
2019-08-21 08:24:28 FINE [org.apache.http.client.protocol.RequestAuthCache] No credentials for preemptive authentication
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalHttpAsyncClient] [exchange: 15] Request connection for {}->http://10.72.10.83:9200
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.PoolingNHttpClientConnectionManager] Connection request: [route: {}->http://10.72.10.83:9200][total kept alive: 1; route allocated: 1 of 10; total allocated: 1 of 30]
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:49236<->10.72.10.83:9200[ACTIVE][r:r]: Set timeout 0
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.PoolingNHttpClientConnectionManager] Connection leased: [id: http-outgoing-0][route: {}->http://10.72.10.83:9200][total kept alive: 0; route allocated: 1 of 10; total allocated: 1 of 30]
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalHttpAsyncClient] [exchange: 15] Connection allocated: CPoolProxy{http-outgoing-0 [ACTIVE]}
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:49236<->10.72.10.83:9200[ACTIVE][r:r]: Set attribute http.nio.exchange-handler
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:49236<->10.72.10.83:9200[ACTIVE][rw:r]: Event set [w]
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE] Request ready
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalHttpAsyncClient] Connection route already established
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 15] Attempt 1 to execute request
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.MainClientExec] Target auth state: UNCHALLENGED
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.MainClientExec] Proxy auth state: UNCHALLENGED
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:49236<->10.72.10.83:9200[ACTIVE][rw:w]: Set timeout 30000
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 >> POST /_search?typed_keys=true&ignore_unavailable=false&expand_wildcards=open&allow_no_indices=true&search_type=query_then_fetch&batched_reduce_size=512 HTTP/1.1
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 >> Content-Length: 326
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 >> Content-Type: application/json
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 >> Host: 10.72.10.83:9200
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 >> Connection: Keep-Alive
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 >> User-Agent: Apache-HttpAsyncClient/4.1.2 (Java/11.0.4)
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:49236<->10.72.10.83:9200[ACTIVE][rw:w]: Event set [w]
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE] Output ready
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 15] produce content
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 15] Request completed
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE] [content length: 326; pos: 326; completed: true]
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:49236<->10.72.10.83:9200[ACTIVE][rw:w]: 647 bytes written
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 >> "POST /_search?typed_keys=true&ignore_unavailable=false&expand_wildcards=open&allow_no_indices=true&search_type=query_then_fetch&batched_reduce_size=512 HTTP/1.1[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 >> "Content-Length: 326[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 >> "Content-Type: application/json[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 >> "Host: 10.72.10.83:9200[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 >> "Connection: Keep-Alive[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 >> "User-Agent: Apache-HttpAsyncClient/4.1.2 (Java/11.0.4)[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 >> "[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 >> "{"size":1000,"query":{"bool":{"must":[{"wildcard":{"config":{"wildcard":"***","boost":1.0}}},{"range":{"message_time":{"from":"2019-08-14 08:24:28.543","to":"2019-08-21 08:24:28.543","include_lower":true,"include_upper":true,"boost":1.0}}}],"adjust_pure_negative":true,"boost":1.0}},"sort":[{"message_time":{"order":"desc"}}]}"
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE] Request ready
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:49236<->10.72.10.83:9200[ACTIVE][r:w]: Event cleared [w]
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:49236<->10.72.10.83:9200[ACTIVE][r:r]: 1038 bytes read
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 << "HTTP/1.1 200 OK[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 << "content-type: application/json; charset=UTF-8[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 << "content-length: 951[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 << "[\r][\n]"
2019-08-21 08:24:28 FINE [org.apache.http.wire] http-outgoing-0 << "{"took":5,"timed_out":false,"_shards":{"total":25,"successful":15,"skipped":0,"failed":10,"failures":[{"shard":0,"index":"accelerator_alarms_config_2019-08-01","node":"rw235SAtTkedi_4NmvMq_g","reason":{"type":"illegal_argument_exception","reason":"Fielddata is disabled on text fields by default. Set fielddata=true on [message_time] in order to load fielddata in memory by uninverting the inverted index. Note that this can however use significant memory. Alternatively use a keyword field instead."}},{"shard":0,"index":"accelerator_alarms_state_2019-08-01","node":"rw235SAtTkedi_4NmvMq_g","reason":{"type":"illegal_argument_exception","reason":"Fielddata is disabled on text fields by default. Set fielddata=true on [message_time] in order to load fielddata in memory by uninverting the inverted index. Note that this can however use significant memory. Alternatively use a keyword field instead."}}]},"hits":{"total":0,"max_score":null,"hits":[]}}"
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 << HTTP/1.1 200 OK
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 << content-type: application/json; charset=UTF-8
2019-08-21 08:24:28 FINE [org.apache.http.headers] http-outgoing-0 << content-length: 951
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE(951)] Response received
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 15] Response received HTTP/1.1 200 OK
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE(951)] Input ready
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 15] Consume content
2019-08-21 08:24:28 FINE [org.apache.http.impl.nio.client.InternalHttpAsyncClient] [exchange: 15] Connection can be kept alive indefinitely
[... cut ...]
org.phoebus.applications.alarm/es_host=10.72.10.83
org.phoebus.applications.alarm/es_port=9200
org.phoebus.applications.alarm/es_index=accelerator_alarms_state_2019-08-01
org.phoebus.applications.alarm/es_max_size=1000

org.phoebus.applications.alarm/server=serv-alarm:9092

org.phoebus.applications.alarm/config_name=Accelerator

org.phoebus.applications.alarm/config_names=Accelerator
`

I am using elasticsearch version 6.8.1.

(Sorry about bad formatting, don't know how to do it better here)
Thanks

shroffk, 2019-08-21:
Hi
Can you try a few things...

1. org.phoebus.applications.alarm/es_index=accelerator_alarms_state_*
2. try to search for "*"...no  pv

MaticPogacnik, 2019-08-21:
Hi, thank you for replay, but this didn't work.

Could the problem be here: 

`
2019-08-21 16:09:04 FINE [org.apache.http.wire] http-outgoing-0 << "{"took":5,"timed_out":false,"_shards":{"total":25,"successful":15,"skipped":0,"failed":10,"failures":[{"shard":0,"index":"accelerator_alarms_config_2019-08-01","node":"rw235SAtTkedi_4NmvMq_g","reason":{"type":"illegal_argument_exception","reason":"Fielddata is disabled on text fields by default. Set fielddata=true on [message_time] in order to load fielddata in memory by uninverting the inverted index. Note that this can however use significant memory. Alternatively use a keyword field instead."}},{"shard":0,"index":"accelerator_alarms_state_2019-08-01","node":"rw235SAtTkedi_4NmvMq_g","reason":{"type":"illegal_argument_exception","reason":"Fielddata is disabled on text fields by default. Set fielddata=true on [message_time] in order to load fielddata in memory by uninverting the inverted index. Note that this can however use significant memory. Alternatively use a keyword field instead."}}]},"hits":{"total":0,"max_score":null,"hits":[]}}"
`
Specifically this part:  `Fielddata is disabled on text fields by default. Set fielddata=true` ?
With which version of elastic search was this tested? It's possible I have incompatible version.

Best regards

kasemir, 2019-08-21:
After starting elasticsearch and before then launching the alarm logger, did you run create_alarm_template.sh to create the alarm templates, 
https://github.com/shroffk/phoebus/blob/master/services/alarm-logger/startup/create_alarm_template.sh ?

You should see the templates like this:
```
lynx -dump http://${es_host}:${es_port}/_template
lynx -dump http://${es_host}:${es_port}/_template/alarms_state_template
```

MaticPogacnik, 2019-08-21:
I used the `create_alarm_index.sh` should I also run `create_alarm_template.sh`?

Here is the output from the commands: https://gist.githubusercontent.com/MaticPogacnik/f9e4320cd3c8a2e0cfc3aba5c9b22150/raw/7153cb23204d928099d525ec66165ae56a948283/es.json 

First one have huge output and second one none.

Thanks!!!

kasemir, 2019-08-21:
You need to use create_alarm_template.sh.
Create_alarm_index.sh was an earlier version that just prepares one index.
Then we learned that if we later want to clean up, for example delete all messages that are 1 month old, the best way is to create one index per day or week.
create_alarm_template.sh creates, well, a template, that will then be used with all these automatically created daily indices.

kasemir, 2019-08-21:
By default, you get monthly indices:
```
# The units of the indices date span: Days (D), Weeks(W), Months(M), Years(Y).
date_span_units=M
date_span_value=1
```
https://github.com/shroffk/phoebus/blob/master/services/alarm-logger/src/main/resources/alarm_logging_service.properties

So you either need to manually create those each month, or apply create_alarm_template.sh.

MaticPogacnik, 2019-08-22:
I think the problem was interactive shell beacuse I was running the service inside the docker in the background.  Which coused the service to exit once it came to the shell. 
I implemented -noshell option and now it works ok. 
Also made pull request #794 . Can I get some feedback please, to merge it into the master?

Best regards

MaticPogacnik, 2019-08-23:
This didn't solve a problem for me.
Does alarm logger provides any detailed logs beside the output it print on the shell at start up?
I am running the same jar with the same start command outside the docker and in inside. But if I am running outside works as expected but inside I am not getting any event to the Elasticsearch.

Docker container can access both Kafka and Elasticsearch.
Tested with:
`root@localhost:/# kafkacat -b 10.72.10.90:9092 -L
Metadata for all topics (from broker -1: 10.72.10.90:9092/bootstrap):
 1 brokers:
  broker 0 at 10.72.10.90:9092
 4 topics:
  topic "Accelerator" with 1 partitions:
    partition 0, leader 0, replicas: 0, isrs: 0
  topic "AcceleratorTalk" with 1 partitions:
    partition 0, leader 0, replicas: 0, isrs: 0
  topic "AcceleratorCommand" with 1 partitions:
    partition 0, leader 0, replicas: 0, isrs: 0
  topic "__consumer_offsets" with 50 partitions:
'
++++++++++++++++++++++++++++++++++++++++++++
'root@localhost:/opt/alarm-logger# curl 10.72.10.83:9200
{
  "name" : "E8h5p8N",
  "cluster_name" : "elasticsearch",
  "cluster_uuid" : "LBOtxTn9TZKDg8qKGUr1MA",
  "version" : {
    "number" : "6.8.1",
    "build_flavor" : "default",
    "build_type" : "docker",
    "build_hash" : "1fad4e1",
    "build_date" : "2019-06-18T13:16:52.517138Z",
    "build_snapshot" : false,
    "lucene_version" : "7.7.0",
    "minimum_wire_compatibility_version" : "5.6.0",
    "minimum_index_compatibility_version" : "5.0.0"
  },
  "tagline" : "You Know, for Search"
}
`

shroffk, 2019-08-23:
If the alarm logger can;t connect to elastic it should be logged

https://github.com/shroffk/phoebus/blob/master/services/alarm-logger/src/main/java/org/phoebus/alarm/logging/ElasticClientHelper.java

I am guessing you have elastic running on a different machine and only the alarm service running on docker.

Can you check the elasticsearch.yml and verify that the configure host and port match the bind host and port.

I am not sure if you already tried it but could you try 'curl -XGET http://my_elastic:9200' from the docker instance

MaticPogacnik, 2019-08-24:
All the services (kafka, es, alarm server, alarm logger) are in the docker.
I made some more investigation and looks like logs are send and saved in the elastic search. I can even query them as:
```
[user@serv-services ~]$ curl -X GET 'http://serv-esservices:9200/accelerator_alarms_state_2019-08-01/_search?format=json&pretty'
{
  "took" : 99,
  "timed_out" : false,
  "_shards" : {
    "total" : 5,
    "successful" : 5,
    "skipped" : 0,
    "failed" : 0
  },
  "hits" : {
    "total" : 240,
    "max_score" : 1.0,
    "hits" : [
      {
        "_index" : "accelerator_alarms_state_2019-08-01",
        "_type" : "alarm",
        "_id" : "mDNXxWwB1Uw9CdxggKl0",
        "_score" : 1.0,
        "_source" : {
          "severity" : "MAJOR",
          "current_message" : "NO_ALARM",
          "mode" : null,
          "message_time" : "2019-08-24 20:35:28.411",
          "current_severity" : "OK",
          "pv" : "test_9_996",
          "latch" : "false",
          "time" : "2019-08-24 20:33:34.408",
          "message" : "LOLO_ALARM",
          "config" : "state:/Accelerator/testComponent/nusano_9_996",
          "value" : "1.0"
        }
      },
      {
        "_index" : "accelerator_alarms_state_2019-08-01",
        "_type" : "alarm",
        "_id" : "nDNXxWwB1Uw9CdxggKnq",
        "_score" : 1.0,
        "_source" : {
          "severity" : "MAJOR",
          "current_message" : "HIGH_ALARM",
          "mode" : null,
          "message_time" : "2019-08-24 20:35:30.410",
          "current_severity" : "MINOR",
          "pv" : "test_9_995",
          "latch" : "false",
          "time" : "2019-08-24 20:33:34.408",
          "message" : "LOLO_ALARM",
          "config" : "state:/Accelerator/testComponent/nusano_9_995",
          "value" : "1.0"
        }
      },
...
```

But they don't show up in phoebus for some reason. This is Phoebus configuration:
```

# --------------------------------------
# Package org.phoebus.applications.alarm
# --------------------------------------

# location of elastic node/s
org.phoebus.applications.alarm/es_host=serv-esservices
org.phoebus.applications.alarm/es_port=9200
org.phoebus.applications.alarm/es_index=accelerator_alarms_state_*
org.phoebus.applications.alarm/es_max_size=1000
[... 65 lines cut ...]
2019-08-24 16:46:12 FINE [org.apache.http.wire] http-outgoing-0 << "content-type: application/json; charset=UTF-8[\r][\n]"
2019-08-24 16:46:12 FINE [org.apache.http.wire] http-outgoing-0 << "content-length: 136[\r][\n]"
2019-08-24 16:46:12 FINE [org.apache.http.wire] http-outgoing-0 << "[\r][\n]"
2019-08-24 16:46:12 FINE [org.apache.http.wire] http-outgoing-0 << "{"took":4,"timed_out":false,"_shards":{"total":10,"successful":10,"skipped":0,"failed":0},"hits":{"total":0,"max_score":null,"hits":[]}}"
2019-08-24 16:46:12 FINE [org.apache.http.headers] http-outgoing-0 << HTTP/1.1 200 OK
2019-08-24 16:46:12 FINE [org.apache.http.headers] http-outgoing-0 << content-type: application/json; charset=UTF-8
2019-08-24 16:46:12 FINE [org.apache.http.headers] http-outgoing-0 << content-length: 136
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE(136)] Response received
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 3] Response received HTTP/1.1 200 OK
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE(136)] Input ready
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 3] Consume content
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.client.InternalHttpAsyncClient] [exchange: 3] Connection can be kept alive indefinitely
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.client.MainClientExec] [exchange: 3] Response processed
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.client.InternalHttpAsyncClient] [exchange: 3] releasing connection
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:57946<->10.72.10.83:9200[ACTIVE][r:r]: Remove attribute http.nio.exchange-handler
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.conn.PoolingNHttpClientConnectionManager] Releasing connection: [id: http-outgoing-0][route: {}->http://10.72.10.83:9200][total kept alive: 0; route allocated: 1 of 10; total allocated: 1 of 30]
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.conn.PoolingNHttpClientConnectionManager] Connection [id: http-outgoing-0][route: {}->http://10.72.10.83:9200] can be kept alive indefinitely
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.conn.ManagedNHttpClientConnectionImpl] http-outgoing-0 10.72.10.60:57946<->10.72.10.83:9200[ACTIVE][r:r]: Set timeout 0
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.conn.PoolingNHttpClientConnectionManager] Connection released: [id: http-outgoing-0][route: {}->http://10.72.10.83:9200][total kept alive: 1; route allocated: 1 of 10; total allocated: 1 of 30]
2019-08-24 16:46:12 FINE [org.elasticsearch.client.RestClient] request [POST http://10.72.10.83:9200/_search?typed_keys=true&ignore_unavailable=false&expand_wildcards=open&allow_no_indices=true&search_type=query_then_fetch&batched_reduce_size=512] returned [HTTP/1.1 200 OK]
2019-08-24 16:46:12 FINE [org.apache.http.impl.nio.client.InternalIODispatch] http-outgoing-0 [ACTIVE] [content length: 136; pos: 136; completed: true]

[user@localhost phoebus-product]$

```

shroffk, 2019-08-25:
hello,

I noticed that the phoebus query is 
```
"message_time":{"from":"2019-08-17 16:46:12.679","to":"2019-08-24 16:46:12.679"
```

while the alarm event times are 

```
"message_time" : "2019-08-24 20:35:28.411",
&
"message_time" : "2019-08-24 20:35:30.410"
```

can you try to expand the start the end times and check...we could then look into

MaticPogacnik, 2019-08-26:
Yes! Thank you!
Looks like all this confusion was about different timezone setting between all this environments.

closed 2019-08-26 by MaticPogacnik

kasemir, 2019-08-26:
We just fixed a related issue with #774 : Time stamps in Kafka messages are UTC, stored in elastic search as UTC. User interface (Phoebus alarm UI, Kibana, ...) convert to local time. Is something missing in the alarm log UI?