ControlSystemStudio/phoebus #818: channel access broken in latest version

kingspride, 2019-09-11:
Hi,
unfortunately I have to report a major issue with phoebus.
the latest build seems to be unable to read any PV.

at first, we tought it was our routing from eduroam wifi, but this is not the case.
any build from before 2019-09-07 works (I create nightlies), every other after that fails to retrieve data from given PVs. maybe the Channelfinder has some issue.
in wireshark, I can see 2 small TCP packages getting transfered to and from my IOC host, and thats it. all the data carrying packages the older versions of phoebus created, are missing.

the latest build from yesterday didnt even compile:

```
[INFO] 
[INFO] ------------------------< org.phoebus:core-pv >-------------------------
[INFO] Building core-pv 4.6.0-SNAPSHOT                                  [11/66]
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ core-pv ---
[INFO] 
[INFO] --- jacoco-maven-plugin:0.8.4:prepare-agent (prepare-agent) @ core-pv ---
[INFO] argLine set to -javaagent:/var/lib/jenkins/.m2/repository/org/jacoco/org.jacoco.agent/0.8.4/org.jacoco.agent-0.8.4-runtime.jar=destfile=/var/lib/jenkins/workspace/Phoebus/core/pv/target/jacoco.exec
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ core-pv ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 5 resources
[INFO] Copying 0 resource
[INFO] 
[INFO] --- maven-compiler-plugin:3.6.2:compile (default-compile) @ core-pv ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 60 source files to /var/lib/jenkins/workspace/Phoebus/core/pv/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] error reading /var/lib/jenkins/.m2/repository/org/epics/epics-util/1.0.1-SNAPSHOT/epics-util-1.0.1-SNAPSHOT.jar; zip END header not found
[...]
```

I hope this can be fixed :)
thanks in advance

kasemir, 2019-09-11:
I can confirm intermittent build errors when using Maven which I saw yesterday because it could not download certain pieces. The affected items changed between `Could not transfer artifact org.apache.commons:commons-compress:jar:1.14 from/to google-maven-central` and `Could not transfer artifact org.epics:epics-pvdata:jar:6.1.3-20190910.135014-5 from/to sonatype-snapshots`, seemed somewhat random.
But I cannot reproduce a fundamental CA-does't-work issue. Code compiled yesterday, which results in a product containing `jca-2.4.3-SNAPSHOT.jar`, runs and connects fine for me.
Kunal might know more about the JCA changes and how they might cause problems in specific scenarios.

kasemir, 2019-09-11:
Looks like you somehow ended up with broken downloads:
```
[ERROR] error reading /var/lib/jenkins/.m2/repository/org/epics/epics-util/1.0.1-SNAPSHOT/epics-util-1.0.1-SNAPSHOT.jar; zip END header not found
[ERROR] error reading /var/lib/jenkins/.m2/repository/org/epics/epics-ntypes/0.3.3-SNAPSHOT/epics-ntypes-0.3.3-SNAPSHOT.jar; zip END header not found
[ERROR] error reading /var/lib/jenkins/.m2/repository/org/epics/epics-pvaccess/5.1.3-SNAPSHOT/epics-pvaccess-5.1.3-SNAPSHOT.jar; zip END header not found
[ERROR] error reading /var/lib/jenkins/.m2/repository/org/epics/vtype/1.0.1-SNAPSHOT/vtype-1.0.1-SNAPSHOT.jar; zip END header not found
```

You can start over by deleting your .m2 directory and thus force Maven to fetch the files again.

Or you can download a working target platform from a successful build (phoebus-target* from https://controlssoftware.sns.ornl.gov/css_phoebus/nightly/) and then use ant to compile without maven trying to improve matters.

kingspride, 2019-09-11:
my jenkins always kills the entire working directory when its finished. next run, it fetches everything from scratch so there shouldn't be any leftovers.

since I'm compiling for all 3 plattforms, I would like build phoebus in completion.
like it worked before 2019-09-07  / -06...

kasemir, 2019-09-11:
> I would like build phoebus in completion.

Well, it does build for me, and it also builds for travis when you check the GitHub commit logs for master or the edm_cvt branch with very recent activity.
You might see local issues best analyzed by somebody who likes and understands maven.

shroffk, 2019-09-11:
Ok... this is not a maven issue

I think I know what has happened is... epicsCoreJava now runs into build issues with openjdk8. This seems to have messed up the nightly snapshot upload which is the cause of the issue.

I just re-uploaded the snapshot builds. Can you try again

kingspride, 2019-09-11:
I wasn't aware that the .m2 directory is in /var/lib/jenkins since thats the jenkins home directory.
hence the working directory would get nuked, but maven still used the cached packages in /var/lib/jenkins/.m2

it does compile now BUT the outcome still wont work as expected :(
when I try to just probe a PV it throws errors and doesnt display the value:
<details>
<summary>error messages on windows</summary>
<p><pre>
java.lang.NoClassDefFoundError: org/epics/vtype/VStringArray
        at org.phoebus.pv.ca.JCA_PV.getMetaData(JCA_PV.java:212)
        at org.phoebus.pv.ca.JCA_PV.connectionChanged(JCA_PV.java:191)
        at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:56)
        at com.cosylab.epics.caj.CAJChannel.setConnectionState(CAJChannel.java:655)
        at com.cosylab.epics.caj.CAJChannel.connectionCompleted(CAJChannel.java:355)
        at com.cosylab.epics.caj.impl.handlers.CreateChannelResponse.internalHandleResponse(CreateChannelResponse.java:48)
        at com.cosylab.epics.caj.impl.handlers.AbstractCAResponseHandler.handleResponse(AbstractCAResponseHandler.java:110)
        at com.cosylab.epics.caj.impl.CAResponseHandler.handleResponse(CAResponseHandler.java:139)
        at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:519)
        at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:431)
        at com.cosylab.epics.caj.impl.CATransport.handleEvent(CATransport.java:369)
        at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.handleEvent(LeaderFollowersHandler.java:77)
        at com.cosylab.epics.caj.impl.reactor.Reactor.processInternal(Reactor.java:400)
        at com.cosylab.epics.caj.impl.reactor.Reactor.process(Reactor.java:284)
        at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.run(LeaderFollowersHandler.java:91)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.ClassNotFoundException: org.epics.vtype.VStringArray
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:583)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
        ... 18 more
java.lang.NoClassDefFoundError: org/epics/vtype/VStringArray
        at org.phoebus.pv.ca.JCA_PV.getMetaData(JCA_PV.java:212)
        at org.phoebus.pv.ca.JCA_PV.connectionChanged(JCA_PV.java:191)
        at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:56)
        at com.cosylab.epics.caj.CAJChannel.setConnectionState(CAJChannel.java:655)
        at com.cosylab.epics.caj.CAJChannel.connectionCompleted(CAJChannel.java:355)
        at com.cosylab.epics.caj.impl.handlers.CreateChannelResponse.internalHandleResponse(CreateChannelResponse.java:48)
        at com.cosylab.epics.caj.impl.handlers.AbstractCAResponseHandler.handleResponse(AbstractCAResponseHandler.java:110)
        at com.cosylab.epics.caj.impl.CAResponseHandler.handleResponse(CAResponseHandler.java:139)
        at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:519)
        at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:431)
        at com.cosylab.epics.caj.impl.CATransport.handleEvent(CATransport.java:369)
        at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.handleEvent(LeaderFollowersHandler.java:77)
        at com.cosylab.epics.caj.impl.reactor.Reactor.processInternal(Reactor.java:400)
        at com.cosylab.epics.caj.impl.reactor.Reactor.process(Reactor.java:284)
        at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.run(LeaderFollowersHandler.java:91)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.ClassNotFoundException: org.epics.vtype.VStringArray
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:583)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
        ... 18 more
java.lang.NoClassDefFoundError: org/epics/vtype/VStringArray
        at org.phoebus.pv.ca.JCA_PV.getMetaData(JCA_PV.java:212)
        at org.phoebus.pv.ca.JCA_PV.connectionChanged(JCA_PV.java:191)
        at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:56)
        at com.cosylab.epics.caj.CAJChannel.setConnectionState(CAJChannel.java:655)
        at com.cosylab.epics.caj.CAJChannel.connectionCompleted(CAJChannel.java:355)
        at com.cosylab.epics.caj.impl.handlers.CreateChannelResponse.internalHandleResponse(CreateChannelResponse.java:48)
        at com.cosylab.epics.caj.impl.handlers.AbstractCAResponseHandler.handleResponse(AbstractCAResponseHandler.java:110)
        at com.cosylab.epics.caj.impl.CAResponseHandler.handleResponse(CAResponseHandler.java:139)
        at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:519)
[... cut ...]
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.ClassNotFoundException: org.epics.vtype.VStringArray
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:583)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
        ... 18 more
</pre>
</p>
</details>

shroffk, 2019-09-11:
hmm... I am guessing that the org.epics.vtype jar is included in the the phoebus lib/ folder

kingspride, 2019-09-11:
should be AWS coretto 11.0.3

shroffk, 2019-09-11:
I rebuilt my product from scratch...
I remove the .m2/repository/.cache
I remove the org/epics folder

Then did a mvn clean verify on the phoebus project

The product was able to build and connected to ca and pva pv's
I feel like you are still getting the old epics.vtype... I have also double checked on sonatype to ensure that the lastest uploaded snapshots are available.

kasemir, 2019-09-11:
When you get messages
```error reading /var/lib/jenkins/.m2/repository/org/epics/epics-util/1.0.1-SNAPSHOT/epics-util-1.0.1-SNAPSHOT.jar; zip END header not found```
you can compare those files with the result of a successful build, checking for example against the `epics-util*jar` in the `phoebus-target*zip` on https://controlssoftware.sns.ornl.gov/css_phoebus/nightly/

pavel-ch, 2019-09-16:
My build is broken now, too. The evil seems to be in the save-and-restore (app and service), which leaks in the org.epics.vtype 1.0.0 which is for example without VStringArray.class etc.
With the save and restore disabled (deleted from poms in app, phoebus-product and services), I am able to build with Maven.

kasemir, 2019-09-17:
Yeah, that's not supposed to happen.
All dependencies need to be listed in https://github.com/shroffk/phoebus/blob/master/dependencies/phoebus-target/pom.xml . It refers to `org.epics ... ${vtype.version}`.

The `save-and-restore/model/pom.xml` then refers to `org.epics` `vtype-json 1.0.0`, which has two problems:

1) Wrong version number
2) It asks for vtype-json, while the dependencies pom that's supposed to define the target platform asks for the more generic vtype module. That may include vtype-json, but who knows.

The very idea of the phoebus dependencies setup is not to depend on maven fetching whatever it feels like today.

georgweiss, 2019-09-17:
I see same issues. It's clearly save-and-restore that refers to vtype-json 1.0.0, that in further down the dependency resolution pulls in vtype version 1.0.0.
I have a local solution I hope to be able to push the next few days.

georgweiss, 2019-09-17:
And by the way... Is vtype-json maintained somewhere, with updated dependencies?

georgweiss, 2019-09-19:
Fix is merged, please confirm that it solves your issues.

kingspride, 2019-09-19:
I will as soon as possible. thank you!

georgweiss, 2019-11-06:
Fixed

closed 2019-11-06 by georgweiss