ControlSystemStudio/phoebus #833: PV access over PVA gateway

georgweiss, 2019-09-25:
Cannot make pv access over a PV gateway work. Using the JCA addr_list setting to specify the gateway address and port. 

Anyone out there with a working configuration?

kasemir, 2019-09-25:
PV Access of course uses the PV access address list, not the CA address list.
See https://github.com/shroffk/phoebus/blob/master/core/pva/src/main/java/org/epics/pva/PVASettings.java  
For example, you need to set either the environment variable or Java property "EPICS_PVA_ADDR_LIST".

kasemir, 2019-09-25:
Those are the same environment variables used by the C++ implementation, but since it's Java, we also allow Java properties, which in this case take precedence over environment variables.
For a production setup, just using the environment vars for both C++ and Java clients is likely easiest.

kasemir, 2019-09-25:
The online doc listed the CA settings, since they're preferences, but not the PVA environment variables, so added them.

georgweiss, 2019-09-26:
I may be wrong here, but setting the GW to is inconsistent between pva (environment variable or java property) and ca (properties setting). Same holds true for CS Studio Eclipse. If you agree on my conclusion I can add the possibility to set pva GW in properties file.

kasemir, 2019-09-26:
Not sure. In the past, the argument has often been:
Every EPICS tool uses environment variables for the EPICS_CA_ADDR_LIST, why does Eclipse use preferences?

So with the new PVA library, we're doing just that: Use the preferences, nothing special to do for CS-Studio.
With CA, you could actually also use environment variables, the underlying JCA. lib will read them. But since we now have a 10 year history of CS-Studio using preferences (and changing the name of those preferences every 3 years, internally actually setting system properties from those preferences to pass them to the underlying JCA lib), that's what we still do for CA in phoebus.

To clean that up, I would argue not to add preferences for PVA, but to remove the preferences for CA, so both CA and PVA use environment variables, the same ones that all the other EPICS tools based on the C++ lib recognize.

georgweiss, 2019-09-26:
I would assume that some users are not familiar with any other EPICS tool besides CS Studio. Also, changing a preference file to me is preferable to changing a start script or login profile to make sure the environment variable is set.
But I will not insist on this, I'm just a bit bothered by the lack of symmetry.

kasemir, 2019-09-26:
> I would assume that some users are not familiar with any other EPICS tool besides CS Studio.

My experience is the other way round: Users at EPICS sites look back at 30 years of experience using EPICS tools, setting EPICS_CA_ADDR_LIST, and then wonder why CS-Studio requires additional configuration steps.

You also keep mentioning "GW settings". These are not settings of the gateway. These are network client settings, where you list the subnets or servers that the client needs to access. Those servers could be IOCs, custom CA/PVA servers for example based on python, or gateways.

georgweiss, 2019-09-26:
OK, my bad, I know it's not only applicable to gateways.
I'm not suggesting to replace usage of an environment variable, only suggesting to make settings consistent between ca and pva.

georgweiss, 2019-11-06:
Will - when time permits - add some code to make things consistent between ca and pva.

closed 2019-11-06 by georgweiss