ControlSystemStudio/phoebus #836: Recursive copy of directory in file browser

georgweiss, 2019-09-26:
If a directory in the file browser is dragged and dropped onto itself or on any other node (directory or file) in the same source directory, a recursive copy operation is  started. The pain that follows is worse on Windows than on Mac OS. 

Will fix it.

kasemir, 2019-09-26:
By default, the file browser copies in response to a drag/drop.
That's because early on I tended to often drag files from the browser to a terminal or text editor, just to get the filename. .. but the file browser then deleted the file, thinking it had been 'moved'.

Compared to other file browsers, the default should be move (which now requires pressing ctrl or shift, don't remember), and the file browser would notice if the drop target didn't do anything, so it would NOT delete files when you drag them into a text editor, terminal, ...

Maybe you can look at that as well.

georgweiss, 2019-09-26:
Sure I can take a look, but this will boost the scope a bit and I may need to push this to after ICALEPCS. If we agree to change the default behavior from copy to move, I would advocate it since - as you suggest - that is the default behavior on Mac, Win and Linux file browsers.

georgweiss, 2019-11-06:
Fixed.

closed 2019-11-06 by georgweiss