ControlSystemStudio/phoebus #845: Core Launcher

shroffk, 2019-10-04:
Currently, the launcher is packaged with the product.
This has consequences on how easily site specific projects can be creates.

It might be best to move the launcher to a stand alone core-launcher module. The products can then include this core module and the other apps and core pieces it needs.

kasemir, 2019-10-04:
Move it into core-ui?

shroffk, 2019-10-04:
Hmm... I think keeping it separate might have some merit.
I don't want to over stuff the core-ui and the core-framework modules

kasemir, 2019-10-04:
Certainly not core-framework since the launcher uses UI, and we have services without UI.
I don't think we'll have products with UI but no launcher, plus it's really just one Launcher class, so I'd be OK with adding that to core-ui instead of creating a single-class module just for the launcher.
Still, since you've already done that with a PR, I'm also OK with merging that PR

closed 2019-10-04 by shroffk