ControlSystemStudio/phoebus #849: Replace Nashorn with Graal JavaScript interpreter?

georgweiss, 2019-10-10:
As Nashorn is tagged as deprecated (but still available in Open JDK 12!), Graal should be investigated as a possible replacement.

georgweiss, 2019-10-10:
Spent some time testing Graal. Results are mixed.

Setting up Graal as the JS interpreter is fairly straight-forward and for the test cases used in my evaluation it seems to run as expected. However, in my tests performance is considerably worse than Nashorn. For the test scripts at hand, Nashorn executes typically faster than 1 ms, while Graal needs 5-10 ms. 

Difference is in how Graal parses - and compiles - JS code. Using Graal this happens in the eval() method of the Graal Context class. So each time a script is to be executed, a call to eval() will parse and compile the JS code. One conclusion from my test is that JS code obtaining references to Java APIs (e.g. PVUtil) are expensive in terms of execution time.

A workaround is to modify the JS script such that a "main" function wraps the code that is to be executed at runtime. This makes it possible to call eval() in the compile() method of the JavaScript support class. Then, in the submit() method a reference to the main function is retrieved and then invoked. In this manner we get execution times comparable to Nashorn (i.e. ~1ms in my test code). The expensive script lines referencing Java APIs must be left outside the wrapper function so they can be evaluated only once when calling eval().

To me the above workaround is maybe not an option as that would require all JS code to be patched, and automated patching would be tricky. So we are left with the alternative to keep in mind which JS constructs that incur a performance hit.

Please comment.

kasemir, 2019-10-10:
I don't see an obvious, perfect solution. On the upside, the situation is not hopeless.
For now, nashorn is still there. Once it gets removed from the JDK, it might appear as a standalone module. If not, looks like Graal would "work". Finally, rhino is actually still there: https://developer.mozilla.org/en-US/docs/Mozilla/Projects/Rhino/Download_Rhino 
For now, doing nothing looks most practical because any change from nashorn would require tweaks to existingg scripts. For rhino, the way to include/import packages to access Java code was different, and you just described what would need to change for Graal.
Once we do need to do something because nashorn really vanishes, we can take a poll among those who actually use JS: Use Graal, or go back to rhino?

georgweiss, 2019-10-10:
Agree. However, I'm not saying that we need to change any scripts if we go to Graal, only that import statements will incur overhead that might hurt a bit.
For the the time being I would also favor to not change and instead deal with it when Nashorn is no longer an option. At that point we could also explore if Graal can be used for the Python scripts, which I suspect it can.
I will archive my attempts so they can be revisited if/when we decide to go for Graal.

georgweiss, 2019-10-11:
For those interested in putting Graal to the test, please checkout branch CSSTUDIO-1115 and run your stuff.

closed 2019-10-14 by georgweiss