ControlSystemStudio/phoebus #850: PVA client subscribe/monitor timeout

charlieking65, 2019-10-10:
Just to give a little context to my question:  I am at Fermilab working on a data pool manager that will allow seamless access to ACNET and EPICS data sources so I'm only using the PVA packages.

When subscribed on a channel, after 30 seconds my subscription times out.   I found that the TCP socket was closing on the remote end thus seeming to be timing out on the IOC.   In an effort to get it working I added `submit(echo_request);` to the end of `checkResponsiveness()`  and that fixed the timeout.  

If we're not going to rely on the the TCP keep alive (which is set on the SocketChannel) then
could the right solution be keeping track of the last send call to the IOC and only send the echo if it's been at least "PVASettings.EPICS_CA_CONN_TMO * 1000 / 2" milliseconds?  The 'idle' value in checkResponsiveness() only pertains to the client receiving from the IOC, not sending.

kasemir, 2019-10-10:
What exactly are you using? Java PVA client with C++ PVA server? Which version of the PVA server?
There have been some iterations on this because for example the original PVA C++ server didn't actually echo any content back for the echo command. The latest PVA C++ server does, and it also requires some traffic.

As for "The 'idle' value in checkResponsiveness() only pertains to the client receiving from the IOC, not sending.", that's exactly right.

The way it should work:
If the server keeps sending packets, for example subscription updates, then `markAlive()` is called which resets the idle time marker. Otherwise, after about half the EPICS_CA_CONN_TMO, the client sends an echo request to the server. Sending the echo request to the server does not reset the timer. In response to the echo request, however, the server is supposed to send a reply, i.e. we'll see a packet from the server, and that resets the idle timer.
That's the logic in https://github.com/shroffk/phoebus/blob/master/core/pva/src/main/java/org/epics/pva/client/ClientTCPHandler.java#L200

> In an effort to get it working I added submit(echo_request); to the end of checkResponsiveness() and that fixed the timeout.

That seems wrong. The code is already submitting an echo request after half the EPICS_CA_CONN_TMO. Should not send extra echo requests more often than that. We also don't send an echo request if we already have items in the send queue, i.e. for some reason we're backed up and cannot send request anyway, so trying to send an echo request would just make matters worse.

So let's try to figure out what's happening in your case:
Whats the client, what's the server, is the echo request not sent after EPICS_CA_CONN_TMO/2? Why not, is the send queue not empty?

charlieking65, 2019-10-10:
I am running R7.0.3  `./softIocPVA -m N='' -d demo.db`

https://github.com/shroffk/phoebus/blob/master/core/pva/src/test/resources/demo.db

The following simple test will  do this

`Oct 10, 2019 4:07:26 PM org.epics.pva.client.PVAClient lambda$0
INFO: 'saw' [CID 1, SID 1 INIT]
Oct 10, 2019 4:07:26 PM org.epics.pva.client.PVAClient lambda$0
INFO: 'saw' [CID 1, SID 1 SEARCHING]
Oct 10, 2019 4:07:26 PM org.epics.pva.client.PVAClient lambda$0
INFO: 'saw' [CID 1, SID 1 FOUND]
Oct 10, 2019 4:07:26 PM org.epics.pva.client.PVAClient lambda$0
INFO: 'saw' [CID 1, SID 1 CONNECTED]
`

after 30 seconds.   It looks like the IOC closes the socket.  And since the connection is reset the monitor is gone.   This will **not** happen for a PV with a scan of 30 seconds because that will allow idle to get above  EPICS_CA_CONN_TMO/2.

```java
import org.epics.pva.data.*;
import org.epics.pva.client.*;
import java.util.concurrent.TimeUnit;

public class Test implements MonitorListener                                                                                                                                                            
{
    public void handleMonitor(PVAChannel channel, java.util.BitSet changes,
                                java.util.BitSet overruns, PVAStructure data)
    {
        System.out.println(data);
    }

    public static void main(String[] args) throws Exception
    {
        PVAClient pva = new PVAClient();
        PVAChannel channel = pva.getChannel("saw");

        channel.connect().get(1, TimeUnit.SECONDS);
        channel.subscribe("", new Test());

        while (true)
            Thread.sleep(1000);
    }
}
```

kasemir, 2019-10-10:
> .. R7.0.3 ..

Ah, turns out that example works fine with 7.0.2.2.
As described above, the client will request an echo if the server sent nothing for half the EPICS_CA_CONN_TMO. When there's still nothing, i.e. no reply to the echo or some other traffic, the client will close the connection.

Looks like in R7.0.3 the server does the same: If the client sent nothing for 30 seconds, the server closes the connection. I'll have to check the 7.0.3 server code to confirm (next week) and add another timer/counter for this. If the client sends an echo in every call to `checkResponsiveness()`, yes, that would also cause the server to reset its timer, but it would do it every 3 seconds and thus add unnecessary traffic. Mid term this will be fixed a different way since we plan to make 'pipelining' the default, so a client that does nothing but receive monitors would periodically update the pipeline window, no need for empty 'echo' messages.

kasemir, 2019-10-10:
OK, added another counter that, in the absence of other client messages, makes the client send 'echo' every 15 seconds. Couldn't check it with R7.0.3, but made sure it still works with 7.0.2.2.

charlieking65, 2019-10-10:
Thanks!  And I wasn't proposing to send the echo every 3 seconds as the solution.  I was just merely showing that it seemed to satisfy the server's timeout since the 15 second timeout wasn't happening.  Sorry, I should've been more clear about that.  Thanks for your quick response to this.

kasemir, 2019-10-10:
Just tried with 7.0.3. As you described, the IOC closes the connection after ~30 secs because the client is silent. With the fix that was just merged, the client sends an echo request every ~15 secs and the connection thus stays open. Thanks for reporting this and the quick replies!

charlieking65, 2019-10-11:
Works perfectly!  Again, thanks for the quick response to this!

closed 2019-10-11 by charlieking65