ControlSystemStudio/phoebus #853: [INFO] Equivalent AbstractWidgetModel

DorianHas, 2019-10-11:
Hi there !

I am integrating widgets to Phoebus project. But some files relies on old Classes that doesn't exist anymore. Is there something like AbstractPVWidgetModel and AbstractWidgetProperty or classes I can rely on  ?

Old file : 
- https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/model/AbstractWidgetModel.java
- https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/properties/AbstractWidgetProperty.java

Dorian

kasemir, 2019-10-11:
For a quick run through the widget's model, representation and runtime classes check the original display builder readme on https://github.com/kasemir/org.csstudio.display.builder#code-overview.

AbstractWidgetModel -> Widget, AbstractWidgetProperty -> WidgetProperty

kasemir, 2019-10-11:
.. and check https://github.com/kasemir/org.csstudio.display.builder#components-of-a-widget  and maybe look at the EllipseWidget or TextUpdate as basic examples.

I'll have to pull that info into the phoebus, i.e. the current version, but the basics of Widget for the model, representation for the SWT graphics and runtime (which often can use the default runtime) remain the same for the Eclipse- and Phoebus-based display builder.

kasemir, 2019-10-12:
The Readme from the Eclipse sources is now in the Phoebus sources,  https://github.com/shroffk/phoebus/tree/master/app/display

kasemir, 2019-11-06:
Can we close this?

closed 2019-11-07 by kasemir