ControlSystemStudio/phoebus #859: channelfinder preferences

shroffk, 2019-10-15:
reported by William Kirstaedter

The current preferences for channelfinder were created to match the ones described in the old java library
However this means that the settings.ini files which use the package name require duplication 
```
org.phoebus.channelfinder/channelfinder.serviceURL=http://fel-cf:8080/ChannelFinder
```

this should be fixed to reduce the need for duplication in the preference path

kasemir, 2019-10-15:
Would it be possible to disable the channel finder, for example by setting an empty URL, when it's not used?

```
org.phoebus.channelfinder/channelfinder.serviceURL=
```

I now either get 'NullPointerException' when setting it to empty or 'Connection refused' with the default localhost.

shroffk, 2019-10-17:
I thought about this a bit when implementing the above request... to me this does not make sense.

We added the module to the product, the SPI tries to register the module, it runs the initialization of the module and reports the warnings above.
If I have a beamline where the url has been misconfigured to be empty then the warning above is very useful.

For the autocomplete, I did add a flag so that if no channelfinder client is available then it does not try to search cf every time a new pv name is being typed.

In my opinion the best way to address Kay's issue would be to maintain a site specific product which only includes the desired app's and modules. This should now be easily possible with the launcher. The SNS product would be not include the channelfinder bits and thus not get the above error.

shroffk, 2019-10-17:
If the question is about the "common product" and what it should include then I feel like this one message might be useful to inform the new users that either they need to configure this or put together their own product without this piece.

closed 2020-06-04 by shroffk