ControlSystemStudio/phoebus #868: Handle unsigned data from channel access

shroffk, 2019-10-18:
channelaccess has no support for unsigned types.
So when area detector images are shipped over channelacess the unsigned data is cast to signed data... this results in incorrect displays.
The solution is to add a formula function which will correct the data coming from ca/jca and put in an the appropriate type container i.e. ArrayUInteger or ArrayULong

shroffk, 2019-10-18:
@kasemir 
I added an extra method `getDescription` to the formula SPI. I hope that is ok

kasemir, 2019-10-18:
getDescription is OK.

The original function took type = [Int8, UInt8, Int16, UInt16, Int32, UInt32, Float32, Float64].

Does it make sense to map an Int32 to a UInt8?
I think the only practical application is to map signed to unsigned or vice versa.

So in the "type" agument, I'd call VTypeHelper.getString(type_parameter) and then look for "U".
So it allows both passing the original enum but also a different [Signed, Unsigned] enum or just a string.
If an U is found, convert the data into unsigned.
If not, convert to signed.

shroffk, 2019-10-18:
The list of types was just copied from area detector specification...

You are right...that there are only 3 cases where the mapping makes sense... in the functions implementation there is a switch case which handles only these three... for the rest it just passes on the data.

I have updated the function to use VTypeHelper...I was planning on adding a new function for String argument for type... but this is surely much better

closed 2019-10-19 by shroffk