ControlSystemStudio/phoebus #869: Handling area detector images in phoebus

shroffk, 2019-10-18:
When trying to connect to ArrayData pv of a sim detector I managed to crash phoebus... phoebus/jca runs out of memory following which NPE are thrown.

Below is an example stack trace.

My current plan is as follows
1. add a check for null in the `DBRHelper` to protect the UI thread
2. find the main cause of the out of memory

```
2019-10-18 10:35:58 WARNING [org.phoebus.pv] 13SIM1:{SimDetector-Cam:1}image1:ArrayData monitor error
java.lang.NullPointerException
	at org.phoebus.pv.ca.DBRHelper.decodeValue(DBRHelper.java:276)
	at org.phoebus.pv.ca.JCA_PV.monitorChanged(JCA_PV.java:351)
	at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:84)
	at com.cosylab.epics.caj.CAJMonitor.monitorChanged(CAJMonitor.java:248)
	at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:155)
	at com.cosylab.epics.caj.impl.requests.EventAddRequest.response(EventAddRequest.java:213)
	at com.cosylab.epics.caj.impl.handlers.EventAddResponse.internalHandleResponse(EventAddResponse.java:61)
	at com.cosylab.epics.caj.impl.handlers.AbstractCAResponseHandler.handleResponse(AbstractCAResponseHandler.java:110)
	at com.cosylab.epics.caj.impl.CAResponseHandler.handleResponse(CAResponseHandler.java:139)
	at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:549)
	at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:431)
	at com.cosylab.epics.caj.impl.CATransport.handleEvent(CATransport.java:369)
	at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.handleEvent(LeaderFollowersHandler.java:77)
	at com.cosylab.epics.caj.impl.reactor.Reactor.processInternal(Reactor.java:404)
	at com.cosylab.epics.caj.impl.reactor.Reactor.process(Reactor.java:288)
	at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.run(LeaderFollowersHandler.java:91)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)

java.lang.NullPointerException
	at org.phoebus.pv.ca.DBRHelper.decodeValue(DBRHelper.java:276)
	at org.phoebus.pv.ca.JCA_PV.monitorChanged(JCA_PV.java:351)
	at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:84)
	at com.cosylab.epics.caj.CAJMonitor.monitorChanged(CAJMonitor.java:248)
	at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:155)
	at com.cosylab.epics.caj.impl.requests.EventAddRequest.response(EventAddRequest.java:213)
	at com.cosylab.epics.caj.impl.handlers.EventAddResponse.internalHandleResponse(EventAddResponse.java:61)
	at com.cosylab.epics.caj.impl.handlers.AbstractCAResponseHandler.handleResponse(AbstractCAResponseHandler.java:110)
	at com.cosylab.epics.caj.impl.CAResponseHandler.handleResponse(CAResponseHandler.java:139)
	at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:549)
	at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:431)
	at com.cosylab.epics.caj.impl.CATransport.handleEvent(CATransport.java:369)
	at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.handleEvent(LeaderFollowersHandler.java:77)
	at com.cosylab.epics.caj.impl.reactor.Reactor.processInternal(Reactor.java:404)
	at com.cosylab.epics.caj.impl.reactor.Reactor.process(Reactor.java:288)
	at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.run(LeaderFollowersHandler.java:91)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
2019-10-18 10:35:58 SEVERE [com.cosylab.epics.caj.impl.DBRDecoder] 
java.lang.OutOfMemoryError: Java heap space
```

kasemir, 2019-10-18:
If I understand correctly, org.phoebus.pv.ca.DBRHelper.decodeValue is called with dbr == null because CAJ hands us that.
In JCA_monitorChanged, we have this:

```
 final CAStatus status = ev.getStatus();
            if (status != null  &&  status.isSuccessful())
            {
                final VType value = DBRHelper.decodeValue(is_array, metadata, ev.getDBR());
```

So does CAJ report the status as OK, but then there's no dbr?
Does it work when you run the JVM with more memory?
How large is the waveform?
Does CAJ no longer check for EPICS CA max array bytes, which we configure via 
org.phoebus.pv.ca/max_array_bytes=100000000

shroffk, 2019-10-18:
> Does it work when you run the JVM with more memory?
I was just testing this... I gave it 2048m and it very quickly hogs it all up and then hits the same error above.

>How large is the waveform?
1024*1024 of Ubyte

kasemir, 2019-10-18:
What's the update rate?
I'll have to look into that. Guess I've concentrated on larger but slower images, which is what we have more commonly here at the beamlines.

shroffk, 2019-10-18:
The update can go up to 60-70hz...it is not fixed

kasemir, 2019-10-21:
I've just started a test setup using the latest area detector release compiled against EPICS 7.0.3.
The default sim detector creates images sized 1024 x 1024, the 13SIM1:cam1:ArrayRate_RBV is close to 200Hz.

By default, the phoebus display manager throttles the image updates to 4Hz because of these defaults:
```
# PV throttle (250)
org.csstudio.display.builder.runtime/update_throttle=250

# Display hesitation to accumulate updates (20)
org.csstudio.display.builder.representation/update_accumulation_time = 20
# Pause after updates (100)
org.csstudio.display.builder.representation/update_delay = 100
# Line Plot update throttle
org.csstudio.display.builder.representation/plot_update_delay = 250
# Image plot update throttle
org.csstudio.display.builder.representation/image_update_delay = 250
```
Setting the Java property `-Dorg.csstudio.javafx.rtplot.update_counter=true` adds an update indicator to the image widget.

Running that in the profiler, I see most of the time spent in CAJ receiving the waveforms.
Since we cannot throttle on the CA server side, we do receive many updates.
On the CA client side, the last value is cached for the thottled updates, so most waveforms are quickly garbage collected. 
Don't see it run out of memory that way.

Maybe there's a scenario where the GC is delayed, and that's when we run out of memory?
What's different in your setup?
My CPU load is very low, which probably means it has time to GC.
Do you have more CPU load, and then the GC is delayed?

shroffk, 2019-10-22:
Ok... So here are some more updates.
I think the problem might be in the probe application.
In the Display Builder screens I am able to use the simulation image both with and without the adData formula up to 170hz without any issues (reasonable cpu and memory load).

When I open the same pv's with probe the heap fills up and we start hitting the out of memory error... at this point it looks like there might be a memory leak in the probe and not it caj.

kasemir, 2019-10-22:
Good find, I'll look at that tomorrow

kasemir, 2019-10-23:
Found issues in Probe. Memory exhaustion was easy to reproduce by simply looking at an area detector image PV like `13SIM1:image1:ArrayData` in probe. Now fixed.

For what it's worth, this is a memory usage history of running a display builder, probe, PV tree on an image.

[image]

Starting at the left, for the first ~2.5 minutes, using Channel Access (13SIM1:image1:ArrayData). Then in the middle section using PVA (pva://13SIM1:Pva1:Image). Then again back to CA.

With PVA, the memory usage sawtooth pattern is slower. Fewer garbage collection runs. With Channel Access, we're using & freeing more memory more rapidly.

The CPU load, not shown, was about the same.

PR #883 (2019-10-23): Probe handling of large arrays
Fixes #869:

Probe ran out of memory when displaying the array value of an image.
Simply formatting the array as a string takes time, and requests to update the UI queued up.

Fix:
 * Probe caches the last value and only schedules a UI update unless that's already pending
 * Format the value string off the UI thread
 * New preference to limit the number of displayed array elements when formatting array as string

While checking PVTree for similar issue, found it's already protected, but treats area detector PV like '13SIM1:cam1:..' as constant, so fixed that.
closed 2019-10-23 by kasemir