ControlSystemStudio/phoebus #878: AuthorizationService - reading conf from the server

pavel-ch, 2019-10-23:
Follow up of #511 and #48
While preserving all features of the current file based authorization, would it be acceptable to enable `AuthorizationService` to read the `authorization.conf` file from a network resource?
Via the current `PhoebusSecurity.authorization_file` preference using http(s):// prefix, checked in the `AuthorizationService.init()` method. All current code paths would remain untouched, only the http(s) check and the following network read steps additionally in.
This small change could solve our demands in this field, hopefully not breaking the security comparing to the file access. Especially this change should not have any impact to those not using the feature.
Thank you for your opinion.

kasemir, 2019-10-23:
I think that's a good idea.
The AuthorizationService already reads the config from a stream, and it already does that in a background thread. For now that stream is based on a local resource/file, but supporting a URL would be fine as well.

pavel-ch, 2019-10-23:
OK Thanks I start an implementation&test. I send a PR after that.

closed 2019-10-25 by kasemir