ControlSystemStudio/phoebus #889: Missing manifest

DorianHas, 2019-10-28:
Hi there,

I build my Phoebus app. Then, I tried to launch the jar directly with the java -jar command. It failed because there is a missing manifest. Do you have an idea of what I am missing ?

[image]

kasemir, 2019-10-28:
You can't just run any odd jar file.
Check the top-level README.md on how to compile and run.

shroffk, 2019-10-28:
https://github.com/shroffk/phoebus#running-the-phoebus-application

Hope this helps...

DorianHas, 2019-10-29:
The fact is I don't have access to the web with the VM. So I can't use mvn or ant to start the app.  That's why I am using java -jar command.

kasemir, 2019-10-29:
Ant doesn't need web access.
Still, you can't just run any odd jar file and hope that it somehow works.
You need to run the proper jar file, and which one that is can be gleamed from the maven config, ant config or the run_maven.sh example:
`(cd phoebus-product/target; java -jar product-*-SNAPSHOT.jar  --add-modules=ALL-SYSTEM)`.

shroffk, 2019-10-30:
If you are building the product on the vm from source you will need the web... even if you are using ant you still have to use maven to get the dependencies.

I better solution would be for you to build the product on a machine where you do have internet access. Bundle the product and transfer it on the vm and run it like the way @kasemir describes in the above comment.