ControlSystemStudio/phoebus #896: Display Builder - small enhancements

pavel-ch, 2019-11-01:
Into this Issue, I will group small contributions in order to enhance Display Builder and its infrastructure. These suggestions appear when we slowly start working on displays.

First issue is about JFX dialogs: on Windows, they show a default icon which is ugly. Setting the Owner to dialogs the icon is inherited from the stage. PR follows.

kasemir, 2019-11-01:
On Windows, how do you launch the program? Something like https://github.com/shroffk/phoebus/blob/master/phoebus-product/phoebus.bat works, and seeing the console messages in the cmd.exe window can be helpful, but it has similar issues to the "ugly icon" that you just fixed for dialogs:
End users tend to not notice the *.bat file. The file ending is by default hidden, and there's a plain batch file icon, no CS-Studio icon. At runtime, you have that extra window, which can be useful to see the console messages, but it's also distracting and has some default icon.

pavel-ch, 2019-11-01:
At the moment, this is not solved. Running the product usually not from bat but from sh file using bash.exe from git installation, effect is the same.
We can create a shortcut .lnk file like this with javaw which does not open command window (tested now):
target: "C:\Program Files\Java\jdk-11.0.4\bin\javaw.exe" -jar product-4.6.0-SNAPSHOT.jar  --add-modules=ALL-SYSTEM
Start in: the installation directory
Icon: we must supply icon in a supported format .ico
[image]

pavel-ch, 2019-11-01:
Shorcut can be generated by powershell on an installation step - to fix the absolute paths inside, see for example:
https://dotnet-helpers.com/powershell/create-shortcuts-on-desktops-using-powershell/

pavel-ch, 2019-11-08:
**Display Builder Open Display Action: New Tab into a named pane**
In my use case, I need a way how to ensure single tab in a named pane (closing all other tabs before opening a new one). I have an implementation in a site specific widget.
I can imagine that this feature can be interesting for others, too.
So what I suggest is to extend the `org.csstudio.display.builder.model.properties.OpenDisplayActionInfo.Target` on SINGLE_TAB enum member, extend the `org.csstudio.display.builder.representation.javafx.ActionsDialog` with this radio button and implement handling in `org.csstudio.display.builder.representation.javafx.JFXStageRepresentation`.
The handling looks principially like this (example from my code, will be simplified):
```
            else if (htarget == HmiTarget.SINGLE_TAB)
            {
                // close all others at target pane
                DockPane targetPane = DockStage.getDockPaneByName(model_widget.propPane().getValue());
                if (targetPane != null)
                {
                    List<DockItem> dockItems = targetPane.getDockItems();
                    if (dockItems != null)
                        for (DockItem dockItem : dockItems)
                        {
                            dockItem.close();
                        }
                }
                // Create Phoebus Target from extended HmiTarget
                Target target = Target.TAB;
                button_action = new OpenDisplayActionInfo("AlarmButtonAction",
                         model_widget.propFile().getValue(),
                         model_widget.propMacros().getValue(),
                         target,
                         model_widget.propPane().getValue());
                handleAction(button_action);
            }
```

Are you interested in this @kasemir ? For me, it is not a big issue not doing this.

kasemir, 2019-11-08:
I can see use cases where you want to keep some section of the display reserved for a certain purpose.
For example, always have a file browser on the left side, and some summary display on the right side.
We can do that by for example creating a layout with 3 sections, placing the file browser and summary display on the left resp. right, then locking them.

I think I understand your use case of replacing the content of a certain display panel, and I think there are already ways to do that without introducing a new 'target' mode.
Lets call it the 'detail' panel. Depending on circumstances, it needs to show 'detail1', 'detail2', ..., one at a time. The 'detail' panel is opened once, then locked, so end users won't accidentally close it or add other tabs to the panel.

Option 1:
A top level 'detail' display contains a navigation tab to show one of detail1, detail2, ... 
The user can thus switch from one detail to another as desired. Only one is shown at a time.

Option 2:
A ring of detail displays. 'detail1' has buttons to replace it with 'detail2', 'detail3' and so on. 'detail2' has buttons to replace it with 'detail1', 'detail3' and so on. The user can again change from one detail to another as desired. Only one is shown at a time. But you have some control over what is available from where. For example, 'detail3' might NOT have a button to open 'detail1' if jumping from 3 to 1 makes no sense for some reason.

Option 3:
If the content of the detail panel should change automatically based on what's happening elsewhere, for example in some central 'main' display, use a PV.
This could be a local PV for just the display, or it could be a real PV like an "operating mode" type of PV. In any case, the detail panel consists of an embedded display widget and a script, triggered by the PV, to set the embedded widget's file name:

detail = PVUtil.getInt(pvs[0])
widget.setPropertyValue("file",  "detail%d" % detail)

pavel-ch, 2019-11-08:
Yes I am aware of all these ways. Each has its specific disadvantages, while the common one is that if there is another application open in the 'detail' panel (imagine probe), this will not close - this was the main reason for implementing the feature.

kasemir, 2019-11-08:
Yes, you'd have to place the 'detail' display in the associated pane and then lock it, so probe etc. won't open there.

pavel-ch, 2019-11-08:
Never mind, maybe my use case is too specific.
I want to allow probe open there (because we have no popup window ...). But want to close it as soon as user changes display...
As stated above I have no problem leaving this extension site specific.

kasemir, 2019-11-08:
I agree that this is quite a specific case. In many scenarios, the 'Option 3' where the content of a locked panel changes based on a PV should work. If you don't want to lock the panel, so probe etc. are also allowed, I think the best solution is a "close other tabs" script that your detail displays call on init. I'll try to come up with an example script.

pavel-ch, 2019-11-08:
Thanks Kay I do not want to waste your time. I can try to do it by myself, good idea. Basically must work, we have only take care not to close ourselves.

kasemir, 2019-11-08:
Note that the context menu of a tab already has a "Close Others" entry. Not sure how easy it is to invoke that from a script.

pavel-ch, 2019-11-08:
Being able to control tabs from scripts would be a great advantage...

kasemir, 2019-11-08:
"Close others" closes all others, i.e. all tabs in that window.
Need to make some more methods public so a script can call them.

kasemir, 2019-11-08:
OK, by adding some helper such a script is now basically this:
```
display = widget.getTopDisplayModel()
this_item = DisplayRuntimeInstance.ofDisplayModel(display).getDockItem()
for item in this_item.getDockPane().getDockItems():
        if item != this_item:
                item.close()
```

.. plus additional code to perform this on the UI thread.

It opens the door to generally interacting with other dockitems in the same dockpane.

pavel-ch, 2019-11-08:
Great, thanks, going to dig into this.

kasemir, 2019-11-08:
The PR contains a complete example script, it's in the Display Builder examples (but not called by any display)

pavel-ch, 2019-11-08:
We mentioned the Probe app in the discussion, so by the way:
Are you thinking about the OPI Probe equivalent in DisplayBuilder, or it is deprecated/not required/... ?
From my view, it would be interesting to have a user / per project defined probe display.
But I understand that having 2 probes (OPI and the real Probe app) can be confusing -> some context menu preference.

kasemir, 2019-11-08:
I can add a preference setting for a "Probe Display" that's invoked with a $(PV) macro.
Sites which configure that might then want to remove the other probe from their product.
Or keep both.

kasemir, 2019-11-08:
OK, there's now a "Probe Display" context menu entry with a default display included in the examples.

pavel-ch, 2019-11-08:
Thank you, works well 👍 🥇 .
I am still thinking about (hopefully not being too annoying) a way how to place this probe display into a specific named pane. Idea is to extend the URI helper `org.phoebus.framework.util.ResourceParser` of the pane=destination_name construct (in the same way like app=app_name). This can enable ALL applications (which support / implement the feature) to create themselves in a named pane, if present. Then, we can simply extend the probe_display preference URI string with a requested destination pane. 
What about that?
_Remark - no need to hurry with implementation. I can do it if agreed._

pavel-ch, 2019-11-26:
Connected with #971, now makes sense the following:
The Display Runtime Context menu command "Open with Editor" availability should be either configurable in settings.ini, or User rights dependent.
I slightly prefer User rights.
Reason: clear separation of development and runtime environments.
Can I implement it?

kasemir, 2019-11-26:
Enabling it based on user rights seems easy:
```
if (AuthorizationService.hasAuthorization("edit_display"))
   // add entries
// else: Don't
```
https://github.com/shroffk/phoebus/blob/4dc16a10e6a50bf6d50e2dcc0c23f67e4aacd918/core/security/src/main/java/org/phoebus/security/authorization/AuthorizationService.java#L90

Could be a little more complicated because some editor support is contributed via SPI, like the Display/New Display menu entry.
It's  easy to fully remove the editor by deleting/not including the app-display-editor JAR in the product. But disabling the SPI entries at runtime.

pavel-ch, 2019-11-26:
I want to avoid not including the app-display-editor JAR in the product, usually it is handy having editor on all installations, but available for authorized users only. In my case, there is no need to remove menu entries because for a pure runtime, the menu is planned to be hidden like #971. Only the Context menu is necessary for me, being easy to handle as you indicated above.
But I can look to other places where to disable editor. Maybe to handle this all in
`org.csstudio.display.builder.editor.app.DisplayEditorApplication.create() / create(final URI resource)`?
A dialog box "Display Editor not available due to missing user rights." can be enough in this case, not bothering with removal or disabling menu entries.

kasemir, 2019-11-26:
Looks like you can do all in the app-display-* code, based on `AuthorizationService.hasAuthorization("edit_display")`, so go ahead.

pavel-ch, 2019-12-18:
We would appreciate a One-Click Save button in DisplayEditor. Editor Toolbar? I checked it and it seems to me not being easy due to the Editor/GUI/Instance objects structure. Any advice (or rejection)? Thanks @kasemir.

kasemir, 2019-12-18:
I'll add the button. Meanwhile, press Ctrl-S or the "Run" button, which saves before running.

pavel-ch, 2019-12-18:
Great, thank you!

pavel-ch, 2019-12-18:
For debugging of PVs in a display, it would be extremely useful to have a list of all PVs in a display together with Connected/Disconnected information. We have the Debug/PV List application, which does only a half of the job - missing Display/Widget information necessary for the display debugging.
I tried to extend WidgetInfoDialog to show all underlying PVs together with widget path for Parent widgets including Display.
See screenshot.
[image]

Can I PR?

kasemir, 2019-12-18:
Does the widget info dialog for a widget only show the PVs on the widget, but when invoked on the display itself, you see all PVs?

pavel-ch, 2019-12-18:
Yes exactly, now I did it also for Group etc showing uderlying PVs but this can be suppressed.

pavel-ch, 2019-12-18:
Simple Widget:
[image]

Group:
[image]

georgweiss, 2019-12-19:
We (ESS) have a very similar request about being able to list disconnected (and connected) PVs at runtime. The proposed change seems to cover this.

pavel-ch, 2020-03-29:
When defining a rule with String values with local characters (like Czech here], these characters are distorted due to Jython handling of Strings:
[image]

In order to fix this, we have to add the 'u' letter before the Jython string, in RuleToString.java line 72:
return "u\"" + escapeString(prop.getValue().toString()) + "\"";
Result:
[image]

Hope this has no interference with others' displays. I will PR, expecting you merge if the change is fine for you.

pavel-ch, 2020-09-04:
When loading a display, there is a time point when the message "Loading Display 'xxx'" appears for a short moment.
This message itself is fine, but:
- there is a flash of scroll bars on the 800*600 position
- there is a flash of background colors between TextArea default and display background.
As a result, loading of a new display is flickering and looks strange.
I see it is introduced in DisplayRuntimeInstance:463 method showMessage, called from loadDisplayFile line 283.
I understand that we do not know the new model size nor background color at that moment.
Is there any idea how to avoid this, @kasemir ? Maybe preferences betting on same background colors? But how to avoid scroll bars?
Thank you.

pavel-ch, 2020-09-09:
I have got a user request for adding the Line Style property to the Polygon Widget (needs a dash-dot for set of polygons). In fact, doing that, it perhaps should be added to Ellipse, Arc and Rectangle, too.
Is that acceptable?

pavel-ch, 2022-03-11:
In our displays, we are using lots of Embedded widgets (in fact site-specific derivatives of Embedded and Symbol). With growing display size, we observe high startup times of displays using 1000s of such widgets. We were able to reduce startup time significantly by introducing a widget local cache, not to read the same small BOB files again and again. The cache stores XML DOM Element tree. To be able to create Display Model from such a cache, we need to add a new constructor to ModelReader class. This has no further consequences in common code. See the PR that follows. Could you approve @kasemir ?

closed 2023-03-14 by pavel-ch