ControlSystemStudio/phoebus #900: core dependencies, specifically between types and logbook

kasemir, 2019-11-06:
The 'core' dependencies currently look like this:

[image]

'types' is quite high in that dependency hierarchy and depends on logbook.
Other tools should be able to use the basic types without getting a logbook dependency, which in turn requires ui.
For example, if I try to add a context menu to the "PV List" tool in UI based on the selected PV, I'd need to add a dependency from ui to types, which creates a circle:

[image]

I suggest changing the dependencies such that logbook uses types, and not the other way around. So the adapter from logbook to PV moves to the logbook module:

[image]

This would allow ui to use types (so that PV list can have a PV-based context menu):

[image]

shroffk, 2019-11-06:
Ok,
I agree with this.

closed 2019-11-06 by kasemir

kasemir, 2019-11-06:
#901