ControlSystemStudio/phoebus #907: Macros inside settings.ini

pavel-ch, 2019-11-07:
Having a multi-project environment (for one user), it would be helpful having a possibility to use macros inside user defined settings.ini file.
Reason: there is a bunch of setting which point to the same directory or the same URL base. Using an external macro based on Java property or Environment variable can help to avoid repeated inputs there. Also the actual user settings.ini path can reasonably help to automatically disinguish between projects.
Example settings.ini in current setup:
```
org.phoebus.hmi.completion/par_file=file://C:/DATA/JavaScript/hmiScada/hmiTemplateProject/par/IEC.PAR
org.phoebus.hmi.pv/svra_url=tcp://localhost:1883
org.phoebus.hmi.display.model.widgets/blink_period=0.5
org.phoebus.hmi.display.model.widgets/mls_file=file://C:/DATA/JavaScript/hmiScada/hmiTemplateProject/res/mls.txt
org.csstudio.display.builder.model/color_files=C:/DATA/JavaScript/hmiScada/hmiTemplateProject/res/color.def
org.csstudio.display.builder.model/font_files=C:/DATA/JavaScript/hmiScada/hmiTemplateProject/res/font.def
org.csstudio.display.builder.model/macros=<RES>C:/DATA/JavaScript/hmiScada/hmiTemplateProject/res</RES><PIC>C:/DATA/JavaScript/hmiScada/hmiTemplateProject/pic</PIC>
org.csstudio.trends.databrowser3/urls=hmi:mysql://localhost/webscada|MySQL
org.csstudio.trends.databrowser3/archives=hmi:mysql://localhost/webscada|MySQL
org.phoebus.hmi.archive.reader/events_url=hmi:mysql://localhost/webscada
org.phoebus.hmi.eventlist.view/filters_file=C:/DATA/JavaScript/hmiScada/hmiTemplateProject/res/AHPFilters.txt
org.csstudio.display.builder.editor/new_display_template=C:/DATA/JavaScript/hmiScada/hmiTemplateProject/res/template.bob
org.phoebus.security/authorization_file=http://localhost:8081/conf/authorization.conf
```
Instead, I can imagine:
```
org.phoebus.hmi.completion/par_file=file://$(settings_location)/par/IEC.PAR
org.phoebus.hmi.pv/svra_url=tcp://$(HMI_SERVER):1883
org.phoebus.hmi.display.model.widgets/blink_period=0.5
org.phoebus.hmi.display.model.widgets/mls_file=file://$(settings_location)/res/mls.txt
org.csstudio.display.builder.model/color_files=$(settings_location)/res/color.def
org.csstudio.display.builder.model/font_files=$(settings_location)/res/font.def
org.csstudio.display.builder.model/macros=<RES>$(settings_location)/res</RES><PIC>$(settings_location)/pic</PIC>
org.csstudio.trends.databrowser3/urls=hmi:mysql://$(HMI_SERVER)/webscada|MySQL
org.csstudio.trends.databrowser3/archives=hmi:mysql://$(HMI_SERVER)/webscada|MySQL
org.phoebus.hmi.archive.reader/events_url=hmi:mysql://$(HMI_SERVER)/webscada
org.phoebus.hmi.eventlist.view/filters_file=$(settings_location)/res/AHPFilters.txt
org.csstudio.display.builder.editor/new_display_template=$(settings_location)/res/template.bob
org.phoebus.security/authorization_file=http://$(HMI_SERVER):8081/conf/authorization.conf
```

I checked the implementation in `org.phoebus.framework.preferences.PropertyPreferenceLoader` and the implementation is possible there, modifying the value which is put to prefs. The only need is to transfer settings file path to the `load` method, modifying all calls to this method from several places in the codebase.

I need a go/nogo for this.
Thanks.

pavel-ch, 2019-11-07:
As to my last considerations, the `$(settings_location)` can be transferred through Java system property, too, because this can be configured in the java command line call in the same way like the settings.ini argument itself (script). So no need to modify `load` method fingerprint.
Basically we find macros in the value string there and replace by a Java system property if available or by an environment variable if available. Otherwise, issue a warning and let the value string as is.
Any opinions @kasemir @shroffk ?

kasemir, 2019-11-07:
Strictly speaking, macros are values set in displays (Action Button, Group, ...) and passed to embedded windows or related displays. But I agree that we can allow Java properties and/or environment variables in settings, in fact `PreferencesReader.replaceProperties` does just that for a few properties. You're basically saying we should always call that by default within the `PreferencesReader`, no need to specifically call it just for a few selected settings.
I think that's a good idea.

kasemir, 2019-11-07:
Can you do this?
1) In `PreferencesReader.replaceProperties`, fall back to checking environment variables if there's no system property.
2) In `PreferencesReader.get`, call `PreferencesReader.replaceProperties`, so we always allow props/env.
3) Remove the now superflous calls to `PreferencesReader.replaceProperties` from outside code, i.e. make it private.

pavel-ch, 2019-11-07:
Ok thanks I look into that and make a prototype.

pavel-ch, 2019-11-07:
The PR follows. The `replaceProperties` has been made not private but package internal due to the PreferencesReaderTest, effect is the same. All external calls changed to simple PreferencesReader.get().

kasemir, 2019-11-07:
Sounds perfect, thanks!

closed 2019-11-07 by pavel-ch

kasemir, 2019-11-07:
Updated the online help, https://github.com/kasemir/phoebus-doc/blob/master/source/preferences.rst