ControlSystemStudio/phoebus #909: Alarm logger setup guidance insufficient

georgweiss, 2019-11-07:
Trying to setup alarm logging following the README in services/alarm-logger. The outcome is a bit disappointing. Here are my observations:
1) Elastic search 7+ does not not work with the create index scripts. 
2) How are the crate index scripts supposed to be used? 
3) When I run create_alarm_index.sh specifying accelerator as argument (does the word "accelerator" have a implied purpose/meaning), an index is apparently created. However, starting the alarm logger server without any arguments, I get the following output full of warnings:
`2019-11-07 12:57:45.846  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_port' was supplied but isn't a known config.
2019-11-07 12:57:45.846  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_port' was supplied but isn't a known config.
2019-11-07 12:57:45.850  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_host' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_host' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_sniff' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_sniff' was supplied but isn't a known config.
2019-11-07 12:57:45.851  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2019-11-07 12:57:45.852  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_sniff' was supplied but isn't a known config.
2019-11-07 12:57:45.915  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_sniff' was supplied but isn't a known config.
2019-11-07 12:57:45.916  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2019-11-07 12:57:45.916  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2019-11-07 12:57:45.949  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2019-11-07 12:57:45.950  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_sniff' was supplied but isn't a known config.
2019-11-07 12:57:45.950  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_sniff' was supplied but isn't a known config.
2019-11-07 12:57:45.950  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2019-11-07 12:57:45.950  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2019-11-07 12:57:45.967  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2019-11-07 12:57:45.967  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2019-11-07 12:57:45.967  WARN 17915 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2019-11-07 12:57:45.967  WARN 17915 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
[... cut ...]
2019-11-07 12:57:46.033  WARN 17915 --- [-StreamThread-1] o.a.k.s.p.i.StreamsPartitionAssignor     : stream-thread [streams-accelerator-alarm-cmd-e0d7c1ac-b59e-4af0-8e6e-99fe3ea48a18-StreamThread-1-consumer] No partitions found for topic acceleratorCommand
`
All alarm related services run on the same host..

The alarm view displays the alarms I am expecting, but the alarm logger apparently is not correctly configured as I do not get any hits for my wildcard query in the alarm log view.

Please provide information on how to debug this and set this up correctly.

kasemir, 2019-11-07:
Delete the create-index scripts. Use the create_alarm_template scripts.
By default, the alarm logger creates a new index every months (settings date_span_units=M, date_span_value=1). So you need an index template for all those indices.

georgweiss, 2019-11-07:
Already tried the template script. That does not make any difference.
The README is in my view incomplete and confusing. I am prepared to updated it based on my effort, but I need input.

kasemir, 2019-11-07:
No idea why you get errors like "The configuration 'date_span_value' was supplied but isn't a known config".

For what it's worth, I run the alarm logger as a systemd service like this under procServ:
```
# File /etc/systemd/system/alarm-logger.service
#
#   sudo cp alarm-logger.service /etc/systemd/system
#   sudo systemctl daemon-reload
#
# Start/status/stop
#   sudo systemctl start alarm-logger.service
#        systemctl status alarm-logger.service
#   sudo systemctl stop alarm-logger.service
#
# Start when computer boots:
#   sudo systemctl enable alarm-logger.service
#
# Log file:
#   less +GF /home/controls/css/alarm-logger/console.log
#
# Console:
#   telnet localhost 4608

[Unit]
Description=Alarm Logger
After=kafka.service

[Service]
User=ky9
Group=slowcontrols_developers
Environment=JAVA_HOME=/home/controls/css/jdk-11
Environment=LOGGER=/home/controls/css/alarm-logger
Environment=PORT=4608
Environment=CONFIG=IHC
Environment=PATH=/home/controls/css/jdk-11/bin:/usr/bin
ExecStart=/usr/bin/procServ --foreground --noautorestart --name alarm-logger --chdir ${LOGGER} --logfile ${LOGGER}/console.log ${PORT} ./alarm-logger.sh -topics ${CONFIG} -date_span_units D -date_span_value 1

[Install]
WantedBy=multi-user.target
```

kasemir, 2019-11-07:
Here are notes from my initial setup, figuring out how to start/stop, how to start over with an empty elastic database, how to check for indices, how to clean up:

[AlarmLogger.txt](https://github.com/shroffk/phoebus/files/3819896/AlarmLogger.txt)

georgweiss, 2019-11-07:
What version of Elasticsearch are you running?

kasemir, 2019-11-07:
elasticsearch-6.3.2

georgweiss, 2019-11-08:
Figured this one out.
I will update the README based on my experience.

closed 2019-11-08 by georgweiss