ControlSystemStudio/phoebus #915: User defined persistecy on PV level

pavel-ch, 2019-11-08:
I am thinking about how to enable users to build a kind of user defined persistence.
I am thinking about creating a new "persisted local" pv type like "ploc://", basically behaving same as loc:// but locally stored. The storage will be at user directory (maybe a preference). Storage is planned as a simple embedded database to store selected VType, still not decided what (as simple as possible) engine to use - advice welcome.
This can be done in my site specific way (SPI) or in core-pv.
Are you interested? (I am planning implementation works mid term, not just now).

kasemir, 2019-11-08:
PVs are often used to control some part of the display, and currently we have two options:
1) Use a local PV, so each user can do their own thing. But when they close/reopen the display, they start over with the default settings.
2) Use a real PV, so all users see the same. Close/reopen the display, and you see that last state.

This would offer a 3rd option: Each user can do their own thing,  and the setup remembers your  last values. Not sure about effort vs. benefit,  but definitely an idea to try as a site-specific addition. Since all PVs use SPI, there'd be no difference if it's later moved into core-pv.

Should store the persisted values in the 'user' location, https://github.com/shroffk/phoebus/blob/master/core/framework/src/main/java/org/phoebus/framework/workbench/Locations.java
The target platform https://github.com/shroffk/phoebus/blob/master/dependencies already contains Derby, an embedded database used by the scan server. So you could use that to store the values. Or a much simpler setup where you create a 'ploc' folder in the user location, with one file per "ploc://" PV.

In any case, my technical concern would be the writing. When implementing PV.write(), https://github.com/shroffk/phoebus/blob/master/core/pv/src/main/java/org/phoebus/pv/PV.java#L212, this should be 'immediate', not blocking until the file or database actually writes. So it needs to remember the most recent value to write, and then write in a background thread.

pavel-ch, 2019-12-19:
Closing, postponing to long term, a possible local use-case disappeared.

closed 2019-12-19 by pavel-ch