ControlSystemStudio/phoebus #920: NPE in save-and-restore because of missing preference defaults

kasemir, 2019-11-08:
When using the `PreferencesReader` with a file like `/save_and_restore_preferences.properties`, the file needs to contain a value (which might be empty) for each preference.

That's in part because the preference file is used to _document_ each available setting, so every setting needs to be listed, ideally with a comment, setting it to the default value.

AppConfig.java:132 tries to read `preferencesReader.get("httpClient.readTimeout")`, but "httpClient.readTimeout" is not listed in the preference file, resulting in

```
java.lang.NullPointerException
	at java.base/java.util.regex.Matcher.getTextLength(Matcher.java:1770)
	at java.base/java.util.regex.Matcher.reset(Matcher.java:416)
	at java.base/java.util.regex.Matcher.<init>(Matcher.java:253)
	at java.base/java.util.regex.Pattern.matcher(Pattern.java:1133)
	at org.phoebus.framework.preferences.PreferencesReader.replaceProperties(PreferencesReader.java:52)
	at org.phoebus.framework.preferences.PreferencesReader.get(PreferencesReader.java:132)
	at org.phoebus.applications.saveandrestore.AppConfig.client(AppConfig.java:132)
```

kasemir, 2019-11-08:
The error is a result of  #907, allowing system properties or environment variables in preferences.

We could update the PreferencesReader to check for null before trying to replace props/env vars.
But in a way I like how it "fails fast" when the preference file doesn't list every setting, with a default value.

In this specific case, the file actually lists the setting, but it's commented:
```
# Read timeout (in ms) used by the Jersey client
# If not set it defaults to 1000 ms
#httpClient.readTimeout=1000
```

Changing it to this would fix the error, defines the default value for sure without requring a comment about it, and is thus shorter:
```
# Read timeout (in ms) used by the Jersey client
httpClient.readTimeout=1000
```

georgweiss, 2019-11-08:
I suggest a null check as there is always a non-zero risk of empty preferences settings.

kasemir, 2019-11-08:
I'll add that with a logged warning.
So people looking at the log output will see that some preference file is missing a setting, but end users will still have a running application.

georgweiss, 2019-11-08:
I'll fix this Monday, if not fixed by then.

kasemir, 2019-11-08:
it's all in #921, just waiting for Travis to check it

closed 2019-11-12 by georgweiss