ControlSystemStudio/phoebus #922: Performance tests for pure java based rules

shroffk, 2019-11-08:
Need a set of tests to confirm the performance of rules implemented using scripts vs the rules that operate in pure java

shroffk, 2019-11-08:
I create a simple test case where the text of label widgets would be changes based on a simple comparison functions  
e.g.  
if ('sim://sine' > 0) -> set text to "greater"
else if ('sim://sine' < 0) -> set text to "lesser"

The above behaviour was implemented using both the old script based rules and the new java rules

for 1000 widgets here is the initial performance comparison

[image]

shroffk, 2019-11-08:
[ScrapPad.zip](https://github.com/shroffk/phoebus/files/3826320/ScrapPad.zip)

georgweiss, 2019-11-11:
For the sake of clarity: in your example bob files, the "old" rules implementation is provided as embedded scripts. Are they the one implicitly created by the "old" rules implementation?

shroffk, 2019-11-13:
I created the old rules using an old version of phoebus...then copied those scripts to the new implementation.
@kasemir I am guessing that this should give us a fair comparison.

georgweiss, 2019-11-13:
The suggested solution - i.e. implementing rules using formula functions - basically means that rules must be converted in bob files, right?

shroffk, 2019-11-13:
yes... I need to complete the conversion aspect of this.
@kasemir would you have time sometime this week to talk about the converters and how to add backward compatibility.

Also, there is another feature that is not supported in the new rules... the old rules were ordered and the processing broke as the first true condition was discovered. These rules work more in a situation were the conditions are mutually exclusive.

I can go over the details of the current implementation and its departure for the old one.

kasemir, 2019-11-13:
Kunal: Yes, that's a good way to compare them.

Georg: What do you mean by "rules must be converted in bob files"?
Ideally, end users don't need to do anything.
Rules work as before, are pretty much added/edited/configured as before.
But there are details:

In BOY, rules were converted into javascript.
In the Display Builder, rules were converted into jython.
Now rules will be handled by core-formula and java code.

If the expression in a rule is "pv0 > 5", that works in all cases.
But with the javascript version, you could write "pv0 > 5  &&  pv0 < 10".
For jython, you need to use "and" instead of "&&".
The rule library patched things like that, but in principle you were free to type any expression that's valid javascript or jython. Now you'd have to type an expression that's handled by the formula.
Based on my experience with the BOY/Display Builder transition, only very few rules used complex expressions, so that's not a big concern.

But I do worry about the ordering. Why can't the new rules handling still respect the order of expressions, check them as listed until there's a match?

georgweiss, 2019-11-13:
Well, in Kunal's example the rules are managed in the display editor using a formula function approach. So the actual rule is persisted that way in the bob file, and is different compared to the current implementation of rules.
Maybe I'm missing something?

kasemir, 2019-11-13:
But end user shouldn't need to convert existing rules, right?

Existing rules look like this:
pv0="sim://sine"
Expr1="pv0 > 5"

In Kunal's update the separate listing of PVs and expressions might go away.
You directly enter
Expr1="`sim://sine` > 5"
But the library needs to read old files and replace the 'pv0' in expressions.

georgweiss, 2019-11-13:
OK, so there is a on-the-fly conversion, missed that one.
Agree, existing rules should not need explicit conversion.

shroffk, 2019-11-13:
> But the library needs to read old files and replace the 'pv0' in expressions.

Yes, and this is something I need to complete along with the conversion from adl and edl screens.
I have already completed the conversion for things like visibility etc...

shroffk, 2019-11-15:
>But I do worry about the ordering. Why can't the new rules handling still respect the order of expressions, check them as listed until there's a match?

I have tried to simplify the rules runner... maybe a hangout might be necessary