ControlSystemStudio/phoebus #928: Inconsistent results when using "Open Display" action

georgweiss, 2019-11-12:
When a button action opens a display, subsequent clicks on the button has varying behavior on different platforms. 
When option "New Tab" is selected:

- Mac + Linux: tab opened by first click is activated.
- Windows: each click opens additional tab.

When option "New Windows" is selected:

- Mac + Linux: window opened by first click is activated, but an additional window with empty content is also opened.
- Windows: each click opens a new window with same (=non-empty) content.

Has anyone noted this? What is the desired behavior for additional button clicks, i.e. new tab/window with same content, or activation of existing?

kasemir, 2019-11-12:
The existing window is located by comparing the input to the `DockItemWithInput`.
If it's found, that existing one is raised.
Otherwise a new one is opened.

That input comparison might fail on windows because of `/` vs `\` in filenames. Would best be investigated by somebody who develops on windows. Fix should then be some 'normalize' step to the inputs before comparing them.

georgweiss, 2019-11-12:
Well, I can investigate on Windows. But what about new windows without content?

kasemir, 2019-11-12:
You'll need to tweak the code that handles the new window. When raising an existing instance, it can skip opening the new window.

georgweiss, 2019-11-12:
I'll look into it. Main thing we agree that additional clicks should reuse an existing window or tab instead of opening duplicated instances.

kasemir, 2019-11-12:
In theory, yes. In practice, this depends on comparing the `DockItemWithInput`'s input, using a plain text comparison. 

Display "x.opi?A=1&B=2"  and display  "x.opi?A=10&B=20" are considered different, even though the display may not actually those macros and thus they're really the same.

Display "x.opi?A=1&B=2"  and display  "x.opi?B=2&A=1" are considered different, even though they're really the same macros and values.

So there can always be duplicates when a screen is opened in various ways. But pushing the same button again should find the existing screen.

georgweiss, 2020-03-19:
This is now a duplicate of #1210

closed 2020-03-19 by georgweiss