ControlSystemStudio/phoebus #929: Query about exporting alarm tree configuration

tanviash, 2019-11-12:
Hi @kasemir , if one needs to export alarm config out of alarm server, is the 'java -jar server.jar -export' the only way to achieve this? Is there something existing in phoebus UI such that users who do not have permissions to execute this command will also be able to get the config.xml ?

kasemir, 2019-11-13:
Why would users not have permission to execute the '-export' command? There's no login etc. required.
Still, unsure why alarm tree UI end users would want to export the config. Those end users who are using the UI tend to react to alarms, maybe disable a nuissance alarm, or update the description of an alarm. They wouldn't  have the first idea what to do with an XML-formatted export file.
More advanced users who perform wider ranging alarm configuration should be able to use the -export command, then load the XML into some text editor. Or they write their own python code to create a new XML file and then import it. So I really don't see a pressing need to add XML import/export to the alarm tree UI.
If you want to implement UI access to the export/import, some of the export/import code would probably need to move from the server to the model package.

shroffk, 2019-11-15:
@tanviash 
The latest alarm config.xml is already maintained in the alarm config service... any config action would result in a new config.xml
As long as the git repo is accessible the user should be able to browse the latest config.xml (I know you were having trouble with this service)

What is the use case you are trying to capture

closed 2019-11-18 by tanviash