ControlSystemStudio/phoebus #940: X/Y Plot - Hiding Traces broken

kingspride, 2019-11-14:
Hi Kunal and Friends,

I have another Bug to report.

I want to hide all traces of an XY Plot  by default and only show them when the user checks a corresponding box.

I therefore disabled the property "visible" on all traces and wrote rules to manipulate it.

The result is: At start of the display, all traces are visible. I have to hide them at runtime.

[image]

[image]

[image]
[image]

kasemir, 2019-11-14:
I can reproduce, will look into it.

kasemir, 2019-11-14:
It's been fixed.

In your screenshot, the PVs used by the checkboxes appear to be invalid until you once toggle them.
If these are records on an IOC, add `field(PINI, "YES")`.
If these are local PVs, provide an initial value like `loc:\\show(0)`.

closed 2019-11-14 by kasemir

kingspride, 2019-11-14:
thanks a lot!

PR #941 (2019-11-14): XYPlot: Initial 'visible' setting for traces
Fixes #940