ControlSystemStudio/phoebus #944: Phoebus release

shroffk, 2019-11-15:
It seems like the product has reached a point where a release would make sense.
The release would also be a good way to co-ordinate with the eclipse based cs-studio release.

georgweiss, 2019-11-15:
Please clarify "coordination" with Eclipse-based CSS. Not in terms of features, I guess, and not as a long-term ambition...?

shroffk, 2019-11-15:
So the Issue_944 has a profile and the release plugin which provides the following functionality.

### Release

#### Step 1:

To perform a release  
```
mvn release:prepare -P release
```
- Check that there are no uncommitted changes in the sources
- Check that there are no SNAPSHOT dependencies
- Change the version in the POMs from x-SNAPSHOT to a new version (you will be prompted for the versions to use)
- Transform the SCM information in the POM to include the final destination of the tag
- Run the project tests against the modified POMs to confirm everything is in working order
- Commit the modified POMs
- Tag the code in the SCM with a version name (this will be prompted for)
- Bump the version in the POMs to a new value y-SNAPSHOT (these values will also be prompted for)
- Commit the modified POMs

Additionally:
- Before commiting the changes, there is a script in the target platform 'release_classpath.py' which will be executed. This script can be modified to updated the .classpath and other files which need manual intervention during a release.

#### Step 2:
Currently, this is not used since we have to resolve where to deploy things to.  
```
mvn release:prepare -P release
```

- Checkout from an SCM URL with optional tag
- Run the predefined Maven goals to release the project (by default, deploy site-deploy)

shroffk, 2019-11-15:
@kasemir the `release_classpath.py` was added to specifically ensure that the commits tagged as release could contain the updated .classpath files which ensure that when a release tagged version is checked out both the maven and ant build work.

georgweiss, 2019-11-15:
ESS offers an Artifactory instance as an SCM deploy target for release builds.

shroffk, 2019-11-15:
@georgweiss @andersharrisson-ess 
You can test this release with 

```
 mvn -P release release:clean -DdryRun=true -DskipTests=true
```

it should produce the final poms and other files without actually changing or pushing

shroffk, 2019-11-15:
It seems like github also got into the deployment business

https://help.github.com/en/github/managing-packages-with-github-packages/using-github-packages-with-your-projects-ecosystem

andersharrisson-ess, 2019-11-18:
Interesting to read about Github Packages. I had no idea they had released a binary repository service. Thank you for the info, @shroffk.
A long term goal could be to use Github Actions and Github Packages to implement CI. Unfortunately Github Packages does not seem to support SNAPSHOT Maven binaries atm, which I would say is a blocker.

georgweiss, 2019-11-18:
If there are no objections, I would like to see a test of the release process as described here (with the additional step to update .classpath and others).

shroffk, 2019-11-18:
Great, 
I have done a test run and it seemed fine.
The 2 steps I have not tested
1. Updating the .classparths
2. Pushing to artifact repository (replease:perform)

@kasemir could you help us with completing the script associated with the ant build

georgweiss, 2019-11-18:
As a short term solution, should we consider ESS Artifactory?

andersharrisson-ess, 2019-11-18:
@shroffk: The release:perfom should be executed by the CI server. This will ensure that we build the releases in a known and isolated environment.
The pipeline will not explicitly run "mvn release:perform", but it will do a mvn deploy on every commit.

georgweiss, 2019-11-19:
@kasemir Please provide input on how to combine the release plugin with your needs to update resources needed for the ant build.
As far as I understand, the release plugin will update poms and commit them without asking questions, right? So running the release goal on the master branch will simply update the master branch, consequently it would be a good idea to do it on your local box using a short-lived branch and then create a pull request.
At ESS our Gitlab CI will start builds on commits and publish them in ESS Artifactory. However, we need to agree that only commits on the master branch would actually trigger build jobs.

kasemir, 2019-11-19:
The maven build creates jar files with version numbers from the pom.xml, while the and build creates jar files with version numbers from the build.xml. If you update one but not the other, it'll still 'work' as in run & compile.

shroffk, 2019-11-26:
@andersharrisson-ess 
It seems like we can use the plugin to push to a branch
http://maven.apache.org/maven-release/maven-release-plugin/examples/branch.html

`mvn release:branch -DbranchName=my-branch -DupdateBranchVersions=true`  

I think we can make a test release on a separate branch to verify behaviour of the above.

closed 2020-02-26 by shroffk