ControlSystemStudio/phoebus #947: Error Log application

pavel-ch, 2019-11-15:
New lines added after app startup are higher than necessary. Platform: Windows 10. See screenshot.
[image]

kasemir, 2019-11-15:
Interesting. I don't see that on Linux. The code doesn't try to do anything about the row height, it just sets the text and for stderr messages the color.
Could this be caused by `\r` in the text? We split lines on `\n`, so that's removed. ... and maybe empty lines should be skipped in the errolog, but that's a different issue.

Can you add code to remove `\r` from the text, like in `LineView.addLine()` to see if some `\r` in the text causes the change in row height?

kasemir, 2019-11-15:
Yes, must be `\r`, because when I add that on Linux I also get higher rows.
Try this in LineDetectingOutputInterpose.java:
```
--- a/app/errlog/src/main/java/org/phoebus/applications/errlog/LineDetectingOutputInterpose.java
+++ b/app/errlog/src/main/java/org/phoebus/applications/errlog/LineDetectingOutputInterpose.java
@@ -71,6 +71,10 @@ public class LineDetectingOutputInterpose extends OutputStream
             line_handler.accept(new String(linebuf, 0, count));
             count = 0;
         }
+        else if (b == '\r')
+        {
+            // Skip Windows line feed
+        }
         else
         {
```

kasemir, 2019-11-15:
Please try the latest on windows, should be OK.

pavel-ch, 2019-11-15:
I can confirm that you were right with the \r on Windows platform, the effect has gone. Hope Mac is ok now, too. Thanks

closed 2019-11-15 by pavel-ch

kasemir, 2019-11-15:
Mac is unhappy because the app is not downloaded from the Apple App Store, but the errlog seems fine.
Thanks for confirming state on Windows!