ControlSystemStudio/phoebus #958: ModelResourceUtil.doResolveResource()

krisztianloki, 2019-11-19:
In #829 the parent display became prioritized over the current working directory (when resolving a relative path). Is there a reason CWD is even considered at all? If it wasn't, then an expensive call to `file.exists()` could be removed or made conditional (if doResolveResource() is called to check if there is an updated file for a legacy resource)

kasemir, 2019-11-19:
CWD is checked last, if the file is not found via other options.
So if your display arrangement doesn't depend on CWD, (and ideally it shouldn't), you never execute that code, and you don't have to worry about its speed.

If you do rely on placing all your files in CWD, then you're also fine most of the time, because a link from some file (in CWD) will first try to resolve relative to that file, i.e. immediately find that file via it's full path constructed from the path of the parent file.

--> In reality, that file.exists test in CWD is very rarely executed.
If you find otherwise, with profiler proof that it's indeed a considerable performance issue, we can reconsider.

krisztianloki, 2019-11-19:
Yes, the file.exists test in CWD is very rarely executed, but the file.exists test to check if it is relative to the parent display will always be executed (to decide if a fallback to CWD is needed) and I cannot really think of a use case where a display arrangement must absolutely needs to depend on CWD.

kasemir, 2019-11-19:
Yes, it checks in a few places until it can resolve the file, because we allow a few different ways of arranging files.
The check is performed in background threads, unlike the original CS-Studio BOY where it was done on the UI thread.
Operationally, I've not seen any issues from the exists() checks.
Can you show a profile run where this is causing a problem?

krisztianloki, 2019-11-19:
I just saw File.exists() showing up in the top 10 of the Hotspot section when I sampled Phoebus opening one of my .bob files. For some reason I cannot attach the screenshot... I will try again tomorrow...

kasemir, 2019-11-19:
Starting the example displays, opening one of the subscreens for a few seconds, close, open another one for a few seconds, close, ... the profiler shows that resolveAndLoadModel uses 0.9% of the CPU, with most of that time spent in actually loading, not resolving/exists() checking:

[image]

krisztianloki, 2019-11-20:
[image]
This is the sampling I got from starting only one of my .bob files (a vacuum screen)

kasemir, 2019-11-20:
What's the call stack, i.e. from where is File.exists called in that example?

krisztianloki, 2019-11-20:
I ran a profiler profiling only `org.csstudio.display.builder.model.util.ModelResourceUtil` and `java.io.File`. If you check the invocation count it is clear that 89% of `File.exists()` calls come from `doResolveResource()`. (I thought it is easier to do it this way; there were too many threads calling `File.exists()`)
[image]

kasemir, 2019-11-20:
Well, I worry that we're trying to optimize something that isn't an operational problem.

When you try to open a *.opi file, we check if maybe there's an updated *.bob version. We really have to check if that file exists.
If you open a *.bob file with relative path, yes, we check if that exists as well, then fall back to trying the CWD.
If you only profile that section, then yes, File.exists() shows up as using most of the time.
But that's the job of resouceResouce, to check various locations until the resource is found.
As you noted, many other pieces of code also call File.exists(), because that's the only way to check if a file exists.

In the bigger picture of opening a file and then running it for at least a minute, the ms spent in File.exists() are negligible, especially since this is off the UI thread.
Why change the way files are resolved, breaking it for those who use CWD, without gaining any operational improvement?

kasemir, 2019-11-20:
I had at a time contemplated to just try and open the resource (URL.openStream, new FileInputStream, ...) so instead of checking then opening, it would open right away, catching the exceptions until it succeeds.
But that has several downsides:

1) Throwing/catching exceptions may be more expensive than checking exists()
2) The caller does need the resolved location, so would have to return a new class to wrap the actual location and the stream, while now resolve returns the location, and other call then the stream
3) File.exists() may take some time when the OS first needs to wake up and check some uncached section of the filesystem. But then the same would have happened with a call to open() on a previously dormant file system. After the OS checked if a file exists, the file system information tends to be cached and the following open() is quite fast.

krisztianloki, 2019-11-20:
I thought that relying on CWD might be some legacy thing no one should use anyway so that it could be removed. I still think that it would be a good idea to at least have a way of disabling that fallback. Relying on CWD can introduce unwanted side-effects; it works on my machine (because I happen to be in the correct directory) but not on yours even if we have the same directory structure and the same .bob files.

kasemir, 2019-11-20:
'CWD' may be a little misleading. It's basically falling back to using the name "as is".
I'm really hesitant to remove that, since it might for example help on Windows with "C:/...." that we don't recognize as absolute, yet it "just works".

krisztianloki, 2019-11-20:
Well, I would say that not recognizing "C:/..." as absolute is a bug :) Anyway, would you accept a solution to create a "preferences setting" (defaulting to keeping the current behavior) to be able to disable this fallback?

kasemir, 2019-11-20:
Sure

kasemir, 2019-11-20:
But note that it still needs to do the job:
When called with *.opi, it needs to check for *.bob (calling File.exists()), and when it returns a file name, that must exist (calling File.exists()). What you can remove is the fallback to CWD, so it will simply not check for that and instead return null (nothing found).

kasemir, 2019-11-20:
. .and that you can just remove, and those people who do use CWD can then learn that it was a bad idea in the first place.

krisztianloki, 2019-11-21:
Looking at the bigger picture this is what I found:
- `doResolveResource()` is private and is only called from `resolveResource()`
- if a file exists then `doResolveResource()` returns its absolute path
- if a file does not exist then `doResolveResource()` returns `null` and then `resolveResource()` returns the resource name unchanged effectively resulting in an implicit fallback to CWD up in the call chain (if the resource was a relative path)

So, there really is no point in checking whether the file exists; *if* not checking whether a *.bob exists in place of an *.opi then the combined absolute path of `parent_display` + `resource_name` could simply be returned without doing any `File.exists()`

kasemir, 2019-11-21:
Feel free to create a branch that simplifies it, I'll then try that here.

closed 2019-11-22 by kasemir