ControlSystemStudio/phoebus #964: XML Question

TimGuiteDiamond, 2019-11-22:
This is not really an issue but just a question about the *.bob xml format.

Most of the properties which appear in the editor appear in the xml, having been lowercased and spaces turned into underscores.

One exception to that is the AlarmBorder, which appears as border_alarm_sensitive

Is there a technical reason for that?

Sorry if this is the wrong place to ask,  perhaps you could link me to the *.bob specification, if such a document exists?

kasemir, 2019-11-22:
The names don't always match 1:1.
For example: "X Position" is just `<x>` in the XML.
`<on_label>` shows as "'On' Label".

`<border_alarm_sensitive>` was the name used in BOY. In general we kept the same XML names.
"Alarm Border" is the human readable name, which can be localized. You're looking at the English version.
I should certainly not show it as "Border Alarm Sensitive", because that seems wrong. "Alarm sensitive border" would be closer to the Queen's English. We've picked "Alarm Border" which is shorter with a meaning that seems to have been obvious to most people for the last 3 years. Megan originally picked it, Claudio kept it when he sorted & reorganized the properties.

While the displayed property names can be localized (https://github.com/shroffk/phoebus/blob/master/core/framework/src/main/java/org/phoebus/framework/nls/NLS.java), the xml tags are mostly set.
Everything has defaults, so just `<widget type="label"/>` is valid, and based on the version info we do look for some of the older (BOY) XML tags. But just as with BOY (or edm or ...) there is no schema that lists them all. You're more than welcome to create that and then maintain it.

TimGuiteDiamond, 2019-11-22:
That makes sense.

So there is a mapping of xml names to human readable names and you can customise this if you wish?

kasemir, 2019-11-22:
The human readable name is defined when defining the property.
Many of the common properties are defined in https://github.com/shroffk/phoebus/blob/master/app/display/model/src/main/java/org/csstudio/display/builder/model/properties/CommonWidgetProperties.java, example:
```
/** 'x' property */
public static final WidgetPropertyDescriptor<Integer> propX =
         newIntegerPropertyDescriptor(WidgetPropertyCategory.POSITION, "x", Messages.WidgetProperties_X,
                    0, Integer.MAX_VALUE);
```

The human readable name is in the localizable messages, see https://github.com/shroffk/phoebus/blob/master/core/framework/src/main/java/org/phoebus/framework/nls/NLS.java.

Admittably, the only localization we have is
1) In a test, https://github.com/shroffk/phoebus/tree/master/core/framework/src/test/resources/org/phoebus/framework/nls
2) @pavel-ch created them for his product, but so far decided not to share them assuming there's not much interest in the language.

closed 2019-11-27 by TimGuiteDiamond