ControlSystemStudio/phoebus #965: *.fxml needs to be in resources

kasemir, 2019-11-22:
Like resources in general, the `*.fxml` files should be under `src/main/resources` and not `src/main/java`.
That way, you can also remove this from the top-level pom:

```
<resource>
  <directory>src/main/java</directory>
  <includes>
    <include>**/*.fxml</include>
  </includes>
</resource>
```

Currenly placed in 'java':
```
./app/email/src/main/java/org/phoebus/applications/email/ui/SimpleCreate.fxml
./app/logbook/ui/src/main/java/org/phoebus/logbook/ui/CreateLogEntry.fxml
./app/logbook/ui/src/main/java/org/phoebus/logbook/ui/LogEntryDisplay.fxml
./app/logbook/ui/src/main/java/org/phoebus/logbook/ui/LogEntryTable.fxml
./app/logbook/ui/src/main/java/org/phoebus/logbook/ui/ListSelection.fxml
./app/logbook/ui/src/main/java/org/phoebus/logbook/ui/LogEntryTableView.fxml
./app/logbook/ui/src/main/java/org/phoebus/logbook/ui/LogEntryCalenderView.fxml
./app/channel/views/src/main/java/org/phoebus/channel/views/ui/ChannelTable.fxml
./app/channel/views/src/main/java/org/phoebus/channel/views/ui/ChannelTree.fxml
./app/channel/views/src/main/java/org/phoebus/channel/views/ui/ListMultiOrderedPicker.fxml
./app/channel/views/src/main/java/org/phoebus/channel/views/ui/SelectEntity.fxml
./app/channel/views/src/main/java/org/phoebus/channel/views/ui/AddProperty.fxml
./app/alarm/logging-ui/src/main/java/org/phoebus/applications/alarm/logging/ui/AlarmLogTable.fxml
./app/diag/src/main/java/org/phoebus/app/diag/ui/DiagView.fxml
./app/save-and-restore/ui/src/main/java/org/phoebus/applications/saveandrestore/ui/snapshot/fxml/SnapshotEditor.fxml
./app/save-and-restore/ui/src/main/java/org/phoebus/applications/saveandrestore/ui/fxml/SaveAndRestoreUI.fxml
./app/save-and-restore/ui/src/main/java/org/phoebus/applications/saveandrestore/ui/fxml/TreeCellGraphic.fxml
./app/save-and-restore/ui/src/main/java/org/phoebus/applications/saveandrestore/ui/fxml/ServiceUnavailable.fxml
./app/save-and-restore/ui/src/main/java/org/phoebus/applications/saveandrestore/ui/saveset/fxml/SaveSetEditor.fxml
```

shroffk, 2019-11-22:
Having them in the same package was a requirement of scenebuilder if I remember correctly

kasemir, 2019-11-22:
They can be in the same package, but as resources:
```
./app/filebrowser/src/main/resources/org/phoebus/applications/filebrowser/FileBrowser.fxml
./app/probe/src/main/resources/org/phoebus/applications/probe/view/ProbeView.fxml
./app/display/representation-javafx/src/main/resources/org/csstudio/display/builder/representation/javafx/WidgetColorPopOver.fxml
```

Placing resources in src/main/resource is a maven convention.
" make sure to put them in src/main/resources and the corresponding package.", https://github.com/edvin/fxlauncher/issues/72.
"Make sure that your sample.fxml is in the src/main/resources/ directory ", https://stackoverflow.com/questions/22000423/javafx-and-maven-nullpointerexception-location-is-required

kasemir, 2019-11-22:
scenebuilder: First Oracle dropped support for it, now Gluon. The last version is for JFX 11, June 2018 https://gluonhq.com/products/scene-builder/

kasemir, 2019-11-22:
Can you check if the scene builder indeed cannot be used with the maven convention?
Maybe there's a way to have it look in additional directories? The last manual I can find is from 2016, though: https://docs.gluonhq.com/scenebuilder/

If scene builder forces us to move the *.fxml into src/main/java, then all *.fxm should be there, which also means that some ant build files need to be changed to copy them from there instead of just using the resources folder.

shroffk, 2019-11-22:
I will try to see if I can configure scenebuilder

shroffk, 2019-12-11:
Ok,
I updated scene builder and was able to resolve the previous issue which required the .fxml files to be in the same folder.
I will fix the pom and move the fxml files to resources.

kasemir, 2019-12-11:
Excellent!
Was there anything special you had to do in scene builder, like somehow configure the resources folder?

shroffk, 2019-12-11:
No...surprisingly
There are configurations to set the fx:controller search path... but I did not do anything other than move things.

I might be missing something and I think I might retest.

shroffk, 2019-12-12:
The only difference is the newer version of scenebuilder... I guess that might be it.

kasemir, 2019-12-12:
Thanks for following up on this!

closed 2019-12-12 by kasemir