ControlSystemStudio/phoebus #966: Unable to get Email Working with Alarm Handler

DanielALS, 2019-11-22:
I've been able to get the email client running in Utilities but not the automated email notification for the alarm handler.

my `settings.ini` has these lines for both the alarm server instance and my Phoebus client.

    org.phoebus.applications.alarm/automated_email_sender=Alarm Notifier <alarm_notifier@host.gov>
    org.phoebus.email.EmailPreferencesi/mailserver=smtp.host.gov

I admit my understanding of email is pretty weak. Please advise what other info I can provide.

shroffk, 2019-11-22:
`org.phoebus.email.EmailPreferencesi/mailserver=smtp.host.gov`

Is this a valid smtp server

DanielALS, 2019-11-22:
yeah, I'm being paranoid....smtp.lbl.gov which works with the Utility email app in Phoebus.

shroffk, 2019-11-22:
I think the preferences are not correct...

https://phoebus-doc.readthedocs.io/en/latest/preference_properties.html#email

Here is the nsls2 example 

https://github.com/shroffk/nsls2-phoebus/blob/master/config/xf23id/settings.ini

DanielALS, 2019-11-23:
That didn't help I'm afraid. I have the stand alone email working with my `settings.ini` so I don't think that was the problem, although I did go ahead and change it. Are there some sanity checks I can do ?
I was able to `echo "testing" | mail user@host` which worked.

DanielALS, 2019-12-02:
Found the log for a particular alarm server I'm testing email with:
`phoebus/services/alarm-server/hb_console.log`:

Last line:
    
    2019-12-02 12:01:10 INFO [org.phoebus.applications.alarm] /HeartBeats/blDiscrete:hb/blDiscrete:hb: Schedule mailto:redacted_username@lbl.gov in 0 s

I get the feeling that `0 s` is bad. Will look for an option to change this.

kasemir, 2019-12-02:
When you configure the mailto:// action, you also configure a delay. Make that more than zero, unless you do want an email right away.
The idea is to have a delay of like 10 minutes:
If there is no operator to acknowledge the alarm in 10 minutes, then send an email.
Otherwise save the electrons and don't send email because operators are already looking into it.

DanielALS, 2019-12-02:
Thanks Kay,
So if email was working, then  should have gotten one already.
I can't find any log saying that the auto mated email failed, or was sent, only scheduled.

kasemir, 2019-12-02:
This is wrong:
```
org.phoebus.email.EmailPreferencesi/mailserver=smtp.host.gov
```

The actual setting is `org.phoebus.email/mailhost`, and `smtp.host.gov` is unlikely valid.
You should have something like this:
```
org.phoebus.email/mailhost=smtp.lbl.gov 
```

See https://phoebus-doc.readthedocs.io/en/latest/preference_properties.html#email

DanielALS, 2019-12-02:
I do have that `org.phoebus.email/mailhost=smtp.lbl.gov ` and the email client from the more generic app under `Utilities`  works.

I'm running the alarm server on test host and the Phoebus client on my local workstation.
When I update the alarm configuration, I'm not so sure the alarm server's config gets updated.

Separate issue, but I think it's confusing me.

kasemir, 2019-12-09:
Looking at this again...
You see this message:
```
2019-12-02 12:01:10 INFO [org.phoebus.applications.alarm] /HeartBeats/blDiscrete:hb/blDiscrete:hb: Schedule mailto:redacted_username@lbl.gov in 0 s
```

What you should see is more like this:
```
2019-12-06 19:48:08 INFO [org.phoebus.applications.alarm] /IHC/BL-2 BASIS: Schedule Trigger Automated email in 900 s
2019-12-06 20:03:08 INFO [org.phoebus.applications.alarm] /IHC/BL-2 BASIS: Send email to [fred@sitel.gov, 1234567890@txt.att.net]
```

Check your configuration. It might have a title of "mailto:..", when in fact the detail that matters isn't set.
On the item that should email, under "Automated Actions",
1) Enter "Send email" for title (exact title doesn't matter)
2) Enter "mailto:fred@site.org" for Detail (mailto:.. matters big time)
3) Enter some delay in seconds. 0 should work, but the idea is to email when there's nobody available to acknowledge the alarm and handle it, so 900 s (15 minutes) is what we use.

kasemir, 2019-12-09:
Adding a warning message, since right now automated actions where the "Detail" doesn't use "mailto:" nor "cmd:" simply don't do anything.

kasemir, 2019-12-13:
@DanielALS can you check your config as described above?
I assume you simply don't have "mailto:..." in the "Detail" of your alarm actions.
You have it in the "Title", as confirmed by the "Schedule..." log message, but that one doesn't really matter. The lack of a following "Send email to .." log message suggests that your "Detail" is empty or wrong, which  was silently ignored but should result in an error message with the latest version of the software.
Please update & check again so we can either close this issue or figure out where to look next.

kasemir, 2020-03-17:
Can we close this?

kasemir, 2020-03-18:
Closing since no reply since December and several sites routinely use "mailto:.." notifications, so seems to work. Re-open if issue still exists.

closed 2020-03-18 by kasemir