ControlSystemStudio/phoebus #968: Silently ignoring widgets with errors

krisztianloki, 2019-11-24:
In `ModelReader.readWidgetsAllowingRetry()` every exception (except `ParseAgainException`) is ignored and reading continues with the next widget. The exceptions are logged (with `WARNING` level) but there are no visual indications to the user that something went wrong and what is displayed is not correct. Moreover, it is really easy to unintentionally delete these "problematic" widgets when the display is saved in the editor.

kasemir, 2019-11-25:
This is by design.
The operator in the control room it's important that the display loads & runs as well as possible under the circumstance. Errors are logged for the system maintainer to investigate, but we don't want to simply stop parsing the display, or prompt with a dialog for every single error.

krisztianloki, 2019-11-25:
What about a single error dialog notifying the operator that there were some errors and they should check the log for more information?
And maybe replacing the unparsed widgets with placeholders (just like the Embedded display and Picture widgets)?

kasemir, 2019-11-25:
Those are both good ideas.

Did you actually run into exceptions while parsing a widget? I don't remember a real-life scenario.
But in any case, assume there's a display with 1000 widgets, it worked OK when it was edited and initially used, but maybe now it's called with some macro that's meant to contain a number, but for some reason contains a string, so 20 of the widgets which use the macro cannot be parsed?!

What we need to avoid:

 * Can't abort loading the file. Still need to show the other 980 widgets, because they might in fact cover all that the operator needs to do right now.
 * Can't open 20 dialog boxes, let alone modal ones, that keep the operator from interacting with the display.

One dialog box would be acceptable, as you suggest with text like "Errors loading file, check log for details". Also OK to put placeholders there, but for a placeholder you do need position, size and ideally widget type, so you can use something like the `UnknownRepresentation` which shows the outline and the widget type.

krisztianloki, 2019-11-25:
It seems that the widgets Claudio created are only available in the ESS build of CS-Studio/Eclipse or were not ported to Phoebus so that's where I encountered those exceptions.
I will try to implement the dialog box first and then play with the placeholders. Thanks for the suggestion of `UnknownRepresentation`!

georgweiss, 2019-11-25:
Yes, there are widgets that are missing in Phoebus when comparing with Display Builder in CS Studio/Eclipse. Currently we are working on porting at least the thumbwheel widget, but there are for sure other candidates. However, porting is complicated by the fact that some of Claudio's widgets have dependencies to third party libraries that we would need to carefully analyze before adding them to Phoebus.
@krisztianloki , can you provide examples of displays using Claudio's widgets?

krisztianloki, 2019-11-25:
@georgweiss , Miklós will show you some of his displays tomorrow.

closed 2019-12-11 by krisztianloki