ControlSystemStudio/phoebus #974: Override logging properties

georgweiss, 2019-11-26:
Currently logging.properties is loaded from classpath resource only. We'd like to be able to override it from command line:
`-Djava.util.logging.config.file=/path/to/logging.properties`

Any objections?

kasemir, 2019-11-26:
What about existing -logging option of launcher, https://github.com/shroffk/phoebus/blob/master/core/launcher/src/main/java/org/phoebus/product/Launcher.java#L73?

georgweiss, 2019-11-26:
Wasn't aware of this, will try it out.
Thanks.

kasemir, 2019-11-26:
Yes, check it out.

The -logging option is handled after Locations.install, after settings have been loaded, ..., during which we might already issue log messages.
It also loads _another_ log config, while we already loaded a default config.

==> Maybe the -logging option should be handled separately, first thing?

Or maybe it's easier to delete the -logging option?

==> Then mention `-Djava.util.logging.config.file` in the `-help` output.

georgweiss, 2019-11-26:
Well, I think it would be clearer in the Luncher code that `-Djava.util.logging.config.file` is attempted first, then classpath resource if there is no file override.

kasemir, 2019-11-26:
I'm OK with removing the -logging option, instead mention `-Djava.util.logging.config.file` in the help.

georgweiss, 2019-11-26:
Actually, I like the current implementation better. I'm assuming here that the second loading of logging settings will override the first, and that the load time is negligible.

kasemir, 2019-11-26:
The JavaDoc says "Reinitialize the logging properties and reread the logging configuration. ", https://docs.oracle.com/javase/8/docs/api/java/util/logging/LogManager.html#readConfiguration

Not sure if that means start over with empty config, or apply settings from file to current settings (which we got from the file in our resources). But in any case, it suggests that it's fine to call that method more than once.

georgweiss, 2019-11-26:
Loading from file seems to overwrite everything loaded from the classpath resource. Tested this by removing one setting in the externa file, and then that was missing in the log config view.

georgweiss, 2019-11-26:
Decided to not change this, we're fine with -logging.

closed 2019-11-27 by georgweiss