ControlSystemStudio/phoebus #982: Product without app-email throwing exception in Display Runtime Context menu support

pavel-ch, 2019-11-30:
Trying to run a product without e-mail support (app-email*.jar not included) generates exception in Display Runtime Context menu support:
`/app-display-runtime/src/main/java/org/csstudio/display/builder/runtime/app/ContextMenuSupport.java` line 201 throws `java.lang.NoClassDefFoundError`.
Maybe there are several other places where the SPI is not considered (like logbook next line).
How to handle this?
A simple way is to surround these statements with `try/catch(Throwable t)` like:
```
        try {
        	items.add(new SendEmailAction(model_parent, "Display Screenshot", display_info, () ->  Screenshot.imageFromNode(model_parent)));
        } catch (Throwable t) {};
        try {
        	items.add(new SendLogbookAction(model_parent, "Display Screenshot", display_info, () ->  Screenshot.imageFromNode(model_parent)));
        } catch (Throwable t) {};
```
Any other suggestions?

kasemir, 2019-12-01:
The email support needs to be included, but it should be possible to disable it.
Right now the `SendEmailAction` disables itself when there's no email support, but it's still shown.
At the time, that was considered a good idea so users see that in principle there could be email support, it's just not enabled.

 If you think it's better to completely hide the email menu entry unless enabled, update the `runtime.ContextmenuSupport` like this:
```
if (EmailPreferences.isEmailSupported())
{
    //... add email actions to context menu...
}
// .. else: Don't show the (disabled) email support
```

Same should then apply to the logbook actions: Fully hide unless supported.

pavel-ch, 2019-12-01:
Yes I think it is better to hide these entries, making the context menu as short and simple as possible. E-mail support has to be configured by a system engineer anyway, no additional value for end user.
PR follows.

closed 2019-12-01 by kasemir