ControlSystemStudio/phoebus #985: New windows (stages) cause memory leak

georgweiss, 2019-12-02:
When detaching a tab a new Stage is created. If the new window - empty or not - is closed it is never garbage collected.
Testing the behavior using a simple (non-Phoebus) code snippet shows that the closed stage (window) is indeed garbage collected as expected.

kasemir, 2019-12-02:
Which class do you check for leakage?
On Linux, I'm looking at "GtkWindow", and they're GCed.

georgweiss, 2019-12-02:
javafx.stage.Stage

kasemir, 2019-12-02:
OK, I see something like this:
Open 2 windows -> 2 Stage instances.
Close 1 window -> Still 2 stage instances.
Open second window -> 2 stage instances.

Open 3 windows -> 3 stage instances.
Close 2 -> Keeping 3 stage instances.
Open 1 (now total of 2 open) -> 2 Stage instances.

So they are not horrendously leaked and never collected. But something is retained for a while, conceptually like a mouse listener remembering the 'last' window that sent an event, then the window is closed and we keep the reference until it sees a new event from a different window.

kasemir, 2019-12-02:
Example: First stage is the visible 'main' stage. Second, closed stage is retained via some 'oldParent', 'oldWindow' chain:

[image]

kasemir, 2019-12-02:
The one in the snapshot above is temporary. Open another 'probe' in the main window, and now the 'oldWindow' Stage gets GCed.

Can't claim that I fully understand this. Seems related to the `CssStyleHelper` and `TabPaneSkin` during layout remembering the 'oldParent' and 'oldWindow' for a while. When then adding/removing tabs in the visible window and thus forcing a new layout run, the old Stage is GCed.

kasemir, 2019-12-02:
For a test, opened two windows with a 'Probe' in each, Window/Save Layout "Two".
Then save a layout "One" with only one window.
Keep changing between these two layouts several times, ending in "One". Force GC -> Profiler shows only one Stage.
So not too worried about a memory leak in our code.
But there is certainly some 'long term memory' in the underlying JavaFX mechanics, the Stage is not GCed right away on close.

georgweiss, 2019-12-02:
Well, I'd say the some Stage objects are not GCed at all, but maybe reused by JavaFX under certain circumstances. In any case, I too have observed that not all creations of a new Stage will end up as non-GCed objects, but I will do some more testing tomorrow.

georgweiss, 2019-12-03:
OK, we need to live with the fact that JavaFX keeps the Stage object around. I was not able to provoke a alarming number of non-GCed objects, after all.

closed 2019-12-03 by georgweiss

kasemir, 2019-12-03:
If you figure out how JFX keeps holding refs, we could 'optimize' this.
See for example `org.csstudio.display.builder.runtime.app.WeakRefWidgetAction` which was added after I noticed that the JFX context menu mechanism has a 'focused' member that kept referencing the last selected context menu item, and if that item was about a widget, we thus kept the complete display model in memory long after a display was closed until somebody opened another context menu ?!

So far, however, this specific case isn't as clear to me, and at the same time it's not consuming enough memory for long enough that it makes my list of things to investigate.