ControlSystemStudio/phoebus #988: Support for CA with EPICS_CA_NAME_SERVERS?

mhedges, 2019-12-03:
I cannot seem to find a way to display EPICS PVs accessible through `EPICS_CA_NAME_SERVERS` in Phoebus. I can see the PVs with the terminal commands like `caget`, `cainfo`, etc after setting up local forwarding to port`5069` and `export EPICS_CA_NAME_SERVERS=localhost:5069`.  However, Phoebus cannot connect to the PVs, even after setting the `addr_list` variable in `settings.ini` to use `localhost:5069`.

Is this not supported, or am I doing something wrong?

kasemir, 2019-12-03:
While caget, caput, .. use the C/C++ implementation of Channel Access, we naturally use the Java implementation, https://github.com/epics-base/jca.

Not sure if it supports the name server idea.
At first glance, it does.

By default, it reads the system property com.cosylab.epics.caj.CAJContext.name_servers or gov.aps.jca.jni.JNIContext.name_servers. We set most of the other properties from Phoebus preferences, but not that one.

https://github.com/epics-base/jca/blob/master/src/core/com/cosylab/epics/caj/CAJContext.java#L465

So try to run phoebus with `-Dcom.cosylab.epics.caj.CAJContext.name_servers=localhost:5069` and `-Dgov.aps.jca.jni.JNIContext.name_servers=localhost:5069`. I think only the first one is needed, but just in case try both.
If that works, we can add a preference setting for it.

mhedges, 2019-12-03:
Thank you for the response.  It looks like those options are ignored.  Here is the output after running `./phoebus.sh -Dcom.cosylab.epics.caj.CAJContext.name_servers=localhost:5069 -Dgov.aps.jca.jni.JNIContext.name_servers=localhost:5069`:

`2019-12-03 13:47:57 WARNING [org.phoebus.ui.application.PhoebusApplication] Ignoring launch parameter '-Dcom.cosylab.epics.caj.CAJContext.name_servers=localhost:5069'
2019-12-03 13:47:57 WARNING [org.phoebus.ui.application.PhoebusApplication] Ignoring launch parameter '-Dgov.aps.jca.jni.JNIContext.name_servers=localhost:5069'`

For reference, here is the version I am using:
`2019-12-03 13:51:13 INFO [org.phoebus.applications.update] Current version  : 2019-12-02 23:59`

mhedges, 2019-12-03:
My mistake!

Running `mvn exec:java -Dcom.cosylab.epics.caj.CAJContext.name_servers=localhost:5069` worked!  I was able to connect a widget to one of my forwarded PVs.

Would it be possible to have a preference setting for this that can be implemented via `./phoebus.sh -settings settings.ini`?  I would be happy to help with this, although I am not much of a java expert, so I would need some help on where to start

kasemir, 2019-12-03:
Would be great if you could try implement this!

1) In https://github.com/shroffk/phoebus/blob/master/core/pv/src/main/resources/pv_ca_preferences.properties, add some new setting, like

```
# Name server list
name_servers=
```

2) In https://github.com/shroffk/phoebus/blob/master/core/pv/src/main/java/org/phoebus/pv/ca/JCA_Preferences.java#L111, check how the 'address list' preference is copied into the system properties used by CAJ. Do the same for the new name_servers

mhedges, 2019-12-04:
Fixed in #989

closed 2019-12-04 by mhedges