ControlSystemStudio/phoebus #991: Display Model / absolute filenames do not work on Windows

pavel-ch, 2019-12-04:
After tweaks in #958 (I guess), absolute resource filenames are not working on Windows.
The requested resource name is resolved into the actual (source) resource path PLUS the requested absolute path.
One possible way to improve the situation and not revert #958 changes is to extend the method:
`org.csstudio.display.builder.model.util.ModelResourceUtil.isAbsolute(String)`, which looks like:
```
    private static boolean isAbsolute(final String path)
    {
        return path.startsWith("/")  ||
               isURL(path);
    }
```
Not evaluating Windows absolute paths like C:/xxx/yyy resp. C:\xxx\yyy just now.
Adding a test to the `':'` at the second position can help in most cases. Hesitating about a test on other variants (UNC paths ..?)
Opinions?

kasemir, 2019-12-04:
I was afraid of changing that mechanism, and actually mentioned potential issues with Windows C:/ in that ticket. Please work with @krisztianloki who wanted this change. I'm OK with adding a check for `[a-zA-Z]:` to test for windows drive letters, or whatever it takes.

krisztianloki, 2019-12-04:
What about using File.isAbsolute() as a third condition, like this?
```java
    private static boolean isAbsolute(final String path)
    {
        return path.startsWith("/")  ||
               isURL(path) || new File(path).isAbsolute();
    }
```

pavel-ch, 2019-12-05:
I see some redundancy, why not:
```
    private static boolean isAbsolute(final String path)
    {
        return new File(path).isAbsolute()  || isURL(path);
    }
```
The `File(path).isAbsolute()` must work consistently on all platforms.

krisztianloki, 2019-12-05:
I left `path.startsWith("/")` there because I thought of it as an optimization

pavel-ch, 2019-12-05:
Right but why not to optimize for Windows then ;-) ? The `new File.isAbsolute()` should be fast as it does not touch the filesystem as what I have read (not investigated too deeply). The costs for the new object construction is included, that is true. But comparing to file or network access should be minimal.
On the other side this is really a minor issue, it only must work as expected. So I add this if you want, but the platform check is redundant here.

pavel-ch, 2019-12-06:
No complains so I leave it as it is.

closed 2019-12-06 by pavel-ch