ControlSystemStudio/phoebus #996: Multiple X axes

georgweiss, 2019-12-05:
I'm comparing the BOY "XY Graph" with the Display Builder "X/Y Plot" with respect to the X axis options. In BOY you can have multiple X axes, while this does not seem to be supported in Display Builder.

Personally I find multiple X axes hard to justify, but since the Display Builder does not support it (correct?), I'd like to know the motivation for that.

kasemir, 2019-12-05:
In the underlying RTPlot I only implemented one X axis. Multiple Y axes, but only one X. never needed more.

georgweiss, 2019-12-05:
Ok, but since I need to deal with complaints, let me put it this way: have you heard of any reqests to support it?

kasemir, 2019-12-05:
I have not had a need to implement it.

kasemir, 2019-12-05:
Let me put it this way:
If I took a survey if somebody might find a use for plots with N horizontal axes, allowing them to be both on the bottom and at the top of the plot, coupled with the option to link axes from different plots, so that moving an axis in one plot accordingly updates the axes in linked plot, I would expect one person to answer: yes.
But operation has not halted for lack of that feature, it would not be trivial to implement, and the added configuration options would complicate the use for everybody else.

georgweiss, 2019-12-06:
@kasemir, you are answering a different question.
The XY Plot has removed a feature available in the BOY equivalent. We have a user lamenting this discrepancy. I'm simply asking:

- If anyone - besides me - has faced complaints.
- The rationale to not port the feature over from BOY, since that decision must actively have been taken at some point. The fact that you "do not need it" does not cover for the case that others need/want it, irrespective of if it can be justified.

closed 2019-12-11 by georgweiss