ControlSystemStudio/phoebus #1000: X11 middle mouse button 'selection' copy/paste

kasemir, 2019-12-09:
User commented that we don't support the middle mouse button 'selection' copy/paste on Linux.

True.

It's fundamentally not supported by JavaFX:
https://bugs.openjdk.java.net/browse/JDK-8088117]
https://stackoverflow.com/questions/30032290/javafx-mouse-clipboard-does-not-work-in-unix/31278321
https://stackoverflow.com/questions/21844255/how-can-i-read-the-x11-clipboard-from-a-javafx-application

It is supported by AWT, and adding this to `ContextMenuPvToClipboard` would publish PV names not only on the normal copy/paste clipboard but also the X11 selection buffer:
```
import java.awt.Toolkit;
import java.awt.datatransfer.ClipboardOwner;
import java.awt.datatransfer.StringSelection;

...
final java.awt.datatransfer.Clipboard x11_sel = Toolkit.getDefaultToolkit().getSystemSelection();
if (x11_sel != null)
    x11_sel.setContents(new StringSelection(text), (board, lost_content) -> {});
```

But while that "works", it's unclear if interfacing with both the AWT and the JavaFX UI this way might have negative side effects. Also doesn't cover our most interesting use case:
1) Operator middle-clicks on some PV in EDM or ... to get it into the X11 selection buffer
2) Operator would like to paste that PV into Data Browser or newly developed display

In those cases, JavaFX text fields which are used to enter the PV name would need to react to middle clicks, which they don't out of the box.

For now leaving this as a Linux-specific issue that's not generally supported, so users need to go via the normal copy/paste (Ctrl-C, Ctrl-V) mechanism.

closed 2019-12-11 by kasemir

kasemir, 2020-01-10:
Fixing the most common use case:
Site uses MEDM or EDM or ... where PV names can usually be copied into the X paste buffer via middle button click. PVs are _only_ copied into the X paste buffer, not the usual clipboard.
Pasting the PVs into Phoebus PV name text fields used to require first pasting them into e.g. an X text editor, then using Ctrl-C to copy into clipboard, finally being able to Ctrl-V paste into Phoebus PV name text field.

On X11/Linux, we now support a plain middle-button-click in Phoebus PV name text fields to paste text from the X11 paste buffer.

This helps when using both old and new tools on Linux, for example in a control room.
Users who remotely access legacy tools on Linux via X servers on Windows and then want to paste PV names from these into Phoebus running on Windows need to use an X server that supports 'syncing' the paste buffer and the clipboard. Same for Mac OS users accessing X tools on remote hosts via ssh and XQuarts (XQuartz->Preference, Pasteboard view, check "Enable syncing").

pavel-ch, 2020-02-19:
Experiencing:
```
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
	at org.csstudio.display.builder.representation.javafx.widgets.RegionBaseRepresentation.hookMiddleButtonCopy(RegionBaseRepresentation.java:230)
```
on Windows.
PR follows.