ControlSystemStudio/phoebus #1006: Alarm Logger : Enable/Disable of alarms showing up as "disabled" in ES

tanviash, 2019-12-10:
Anytime I change the enable, latch or annunciate configuration options for a PV, its value in log in the ElasticSearch shows up as false always. 

{
        "_index" : "alarm_rea_alarms_config_2019-12-01",
        "_type" : "alarm_config",
        "_id" : "5ciE8W4BizTjUFf_aDrr",
        "_score" : 1.0,
        "_source" : {
          "config_msg" : "AlarmConfigMessage [user=ashwarya, host=csstudio-dev, description=sim://ramp, delay=0, count=0, filter=null, guidance=null, displays=null, commands=null, actions=null, delete=null]",
          "message_time" : "2019-12-10 20:33:36.964",
          "host" : "csstudio-dev",
          "config" : "config:/ALARM_REA/test-node-2/sim://ramp",
          "user" : "ashwarya",
          "enabled" : "false", <== Should have been true
          "latching" : "false"  <== Should have been true
        }
      },

kasemir, 2019-12-11:
The JSON messages are minimized by using defaults.
A config update defaults to enabled: true, latching: true, annunciating: true, delay:0, count:0, no filter.

So the alarm config logger needs to apply these defaults. When a message as copied in the "config_msg" has no enabled, latching, annunciating element, those should all be true.

shroffk, 2019-12-11:
hmm...
Enable changes from false --> true... but we send nothing and expect the client to understand nothing == true

kasemir, 2019-12-11:
> we send nothing

We send config updates.

Disabled:
```
CreateTime:1576075754880: config:/IHC/...SomePV: {"user":..."enabled":false,..}
```

Enabled:
```
CreateTime:1576075757419: config:/IHC/...SomePV: {"user":"ky9",..}
```

When disabled, we mention "enabled:false". If we don't mention enabled, the default is true.

> user having to cache the previous values

There is no caching required. All the config updates are "complete", because Kafka compaction only leaves us with the last message. So each config message must be complete, it's not an incremental modification of the last message.
But the messages are optimized by assuming defaults: enabled, latching, annunciating default to true, delay to 0, displays to none, ...

tanviash, 2019-12-11:
Whether I disable or enable a PV, I don't see flags for enable, annunciate or latch in the config_msg that's printed out in ES log.

For instance, this is the config_msg from ES log that got disabled.
"config_msg" : "AlarmConfigMessage [user=ashwarya, host=csstudio-dev, description=sim://sawtooth, delay=0, count=0, filter=null, guidance=null, displays=null, commands=null, actions=null, delete=null]",

kasemir, 2019-12-11:
Then the message parser used by the alarm logger must be broken.
I copied the messages from monitoring the kafka stream. It includes `"enabled":false`, and both the alarm server and the UI client parse it correctly.

kasemir, 2019-12-11:
AlarmConfigMessage.toString is what you see in the config_msg from ES.
It forgot to include enabled, latching, ...

kasemir, 2019-12-11:
https://github.com/shroffk/phoebus/blob/master/app/alarm/model/src/main/java/org/phoebus/applications/alarm/messages/AlarmConfigMessage.java#L186

shroffk, 2019-12-11:
I am updating this

shroffk, 2019-12-11:
I am working on this.
The AlarmConfigMessage is deprecated
The AlarmMessage is the one that needs updating since we merged config+state

tanviash, 2019-12-11:
Thanks @shroffk @kasemir

kasemir, 2019-12-11:
AlarmMessage.AlarmStateJsonMessage seems wrong. A "state" message doesn't contain enabled, annunciating, delay, count, filter. Those are config messages for a leaf node (PV).
JsonModelWriter and JSonModelReader shows what elements are written and read for each message.

shroffk, 2019-12-11:
AlarmMessage.AlarmStateJsonMessage actually marks those fields as ignore

shroffk, 2019-12-11:
Ok, I think I have it working, the solution was rather simple. I explicitly added the defaults in the message pojos.

@tanviash can you please test the branch Issue_1006 and verify the fix.

```
            {
                "_index": "accelerator_alarms_config_2019-12-01",
                "_type": "alarm_config",
                "_id": "SucG9m4B341g0Pyvw0jK",
                "_score": 1,
                "_source": {
                    "config_msg": "AlarmConfigMessage [user=kunal, host=DESKTOP-234GNQN, description=loc://test, delay=4, count=0, filter=null, guidance=null, displays=null, commands=null, actions=null, delete=null]",
                    "message_time": "2019-12-11 17:34:28.783",
                    "host": "DESKTOP-234GNQN",
                    "config": "config:/Accelerator/Test/node/loc://test",
                    "user": "kunal",
                    "enabled": "true",
                    "latching": "true"
                }
            },
            {
                "_index": "accelerator_alarms_config_2019-12-01",
                "_type": "alarm_config",
                "_id": "SecG9m4B341g0PyvZ0gw",
                "_score": 1,
                "_source": {
                    "config_msg": "AlarmConfigMessage [user=kunal, host=DESKTOP-234GNQN, description=loc://test, delay=4, count=0, filter=null, guidance=null, displays=null, commands=null, actions=null, delete=null]",
                    "message_time": "2019-12-11 17:34:04.416",
                    "host": "DESKTOP-234GNQN",
                    "config": "config:/Accelerator/Test/node/loc://test",
                    "user": "kunal",
                    "enabled": "false",
                    "latching": "true"
                }
            }
```

tanviash, 2019-12-11:
@shroffk Tested Issue_1006 branch. Fix works.

shroffk, 2019-12-11:
@tanviash I also did a bit of code cleanup

- removed the unused state and config loggers
- merged the properties files

shroffk, 2019-12-11:
You will have to make sure that your startup scripts etc... might have to be updated.
If you are using the cmd like -settings then I think everything should work as before.

tanviash, 2019-12-11:
@shroffk Do we need to change the message_time timestamp to use system default in alarm logger? It currently puts a UTC timestamp in ES.

kasemir, 2019-12-11:
I think that's desired. Check commits a8f91337fbc77b0474957fc0f9d373c496d7e49a and 31bf5282da855deea4f3be5022b1c16fa810f765, we specifically sent UTC time stamps. The display (like Kibana) then converts to local time. The data is stored as UTC.

kasemir, 2019-12-11:
#774 was the pull request. Before using UTC in ES, the Kibana displays were all 5..6 hours off because it assumes UTC in the data.

closed 2019-12-11 by tanviash