ControlSystemStudio/phoebus #1007: IndexOutOfBound - Update failed

DorianHas, 2019-12-11:
Hello there,

I have an issue when using the Databrowser. I assume the buffer of live values doesn't flush itself. Live buffer is setted to 5000 and automatic_history_refresh to true but it doesn't change anything. So every 15 or 40 minutes (depending on the number of values) a displayed PV disappear from the Databrowser with this error :

```
2019-12-10 15:52:18 WARNING [org.phoebus.ui.application.PhoebusApplication] Update failed
java.lang.IndexOutOfBoundsException: Index: 18160
        at java.base/java.util.Collections$EmptyList.get(Collections.java:4483)
        at org.csstudio.trends.databrowser3.model.PlotSampleArray.get(PlotSampleArray.java:37)
        at org.csstudio.trends.databrowser3.model.PlotSampleArray.get(PlotSampleArray.java:16)
        at org.csstudio.javafx.rtplot.internal.TracePainter.drawValueStaircase(TracePainter.java:254)
        at org.csstudio.javafx.rtplot.internal.TracePainter.paint(TracePainter.java:136)
        at org.csstudio.javafx.rtplot.internal.Plot.updateImageBuffer(Plot.java:685)
        at org.csstudio.javafx.rtplot.internal.PlotCanvasBase.lambda$new$1(PlotCanvasBase.java:207)
        at org.phoebus.ui.javafx.UpdateThrottle.lambda$new$0(UpdateThrottle.java:92)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:834)

2019-12-10 15:52:19 SEVERE [org.phoebus.ui.application.PhoebusApplication] UI Updates resume
```

My config : 
- 4 core x 4 sockets
- 16Go RAM
- CentOs 7
- Java 11

Regards,
Dorian

kasemir, 2019-12-11:
Is there archived data for that channel?
Does that channel originally fetch archived data?
Are there other data browser errors before you get this java.lang.IndexOutOfBoundsException?

DorianHas, 2019-12-11:
There is archive and live data for that channel. It gathers both. I have four PLTs running at the same time. Two have 7 channels and two others have 12 channels.

[image]

There are no errors before this one. As you can see on the picture above, channels are not displayed. Each undisplayed channel correspond to an error.

[image]

kasemir, 2019-12-11:
Well, at first glance the automatic_history_refresh works fine for me.
I'll look around and might come up with some debug statements for you to add to your code so we see where the plot samples are reset.

Meanwhile, reduce the live buffer size to something that causes the automatic_history_refresh to run more often. Like 10 samples for a channel that updates at 1 Hz, so it'll run every 10 seconds.
Do you get this error when the automatic_history_refresh runs for the first time? Or does it work a few times, then fails?

DorianHas, 2019-12-11:
I very don't know. The only thing I can tell you is thaht this issue shows up after 15 to 40 minutes.  Can the buffer be overwhelm and cause the error ?

kasemir, 2019-12-11:
For the channels that disappear, I assume if you open the "Inspect Samples" panel and look at its samples, there are none?
Again set the live buffer size so something smaller and see if that makes the issue happen sooner or more often.

kasemir, 2019-12-11:
Not sure how to explain your stack traces:
```
 at org.csstudio.trends.databrowser3.model.PlotSampleArray.get(PlotSampleArray.java:16)
 at org.csstudio.javafx.rtplot.internal.TracePainter.drawValueStaircase(TracePainter.java:254)
```
```
at org.csstudio.trends.databrowser3.model.PlotSampleArray.get(PlotSampleArray.java:16)
at org.csstudio.javafx.rtplot.internal.TracePainter.drawMinMaxArea(TracePainter.java:339)
```

They show that the trace data is of type `PlotSampleArray`.
But the data of PVs is of type `PVSamples`.
When I set a breakpoint at those TracePainter locations, the call to `data.get(i)` ends up in `PVSamples.get()`, not `PlotSampleArray.get()`.

Are those traces all Formulas, not PVs?

kasemir, 2019-12-11:
Just found an issue with the locking of formulas. They internally `synchronize`, but the rest of the code uses the sample's read/write lock.
In case your traces are based on formula items, that would explain it.
The `databrowser_debug_1007` branch should fix that problem.

DorianHas, 2019-12-12:
I can read PVs from command line caget or by inspecting samples. They are just not displayed. I also used the application Probe to validate that Phoebus can access and read PVs even if they were  not displayed.

There are 7 or 12 PVs and a dozen formulas per PLT.

Phoebus has been running for 10 hours now. The issue didn't show up again. Changing the live buffer size may have solved the issue ( 5000 => 2500 ). I will reincrease the buffer size to confirm this is a live buffer issue.

DorianHas, 2019-12-12:
I take back what I just said ...

[image]

Changing the live buffer size delayed the issue.

kasemir, 2019-12-12:
> .. formulas ..

OK, so as suspected, the issue is related to the locking of formulas. They internally synchronize, but the rest of the code uses the sample's read/write lock.
Try running the latest  code, including the fixes from #1012.

DorianHas, 2019-12-12:
Unfortunatly, the issue is still showing up.

[image]

Can it be the values buffer for formulas that causes issues ?

kasemir, 2019-12-12:
Can you attach the *.plt file where you see that error?

Are you sure that you have compiled from the latest sources? In `FormulaItem`, `compute()` should contain
```
if (! samples.lock.isWriteLockedByCurrentThread())
         logger.log(Level.WARNING, "Samples for " + getName() + " are not locked! " ...
```

kasemir, 2019-12-12:
> Can it be the values buffer for formulas that causes issues ?

Absolutely, but not sure why/how.

PV items hold their samples in `PVSamples`, which consist of a section for live PV data and one for archived data. It's affected by the buffer size and the automatic_history_refresh.

Formula items hold their samples in a `PlotSampleArray`. It's size is based on the values in the formula inputs. I've fixed one locking problem, and just added logging in case these sample buffers are accessed without holding their lock.

Please rebuild with the latest code from git 'master' and see if there are error messages.

DorianHas, 2019-12-13:
Can my settings.ini file help ?
[Regul_plt.zip](https://github.com/shroffk/phoebus/files/3959929/Regul_plt.zip)

I've done it this morning I am waiting for the error to show up. I'll keep you updated.

kasemir, 2019-12-13:
You're not running  the latest code.
Your exception stack traces show a call to PlotSampleArray.get() in PlotSampleArray.java line 37.

In the latest code, the 'get' moved to PlotSampleArray.java line 45:
https://github.com/shroffk/phoebus/blob/master/app/databrowser/src/main/java/org/csstudio/trends/databrowser3/model/PlotSampleArray.java#L45

--> I would expect you to see the error in the old code.
In the new code, that exception would look different and ideally not happen at all.

DorianHas, 2019-12-16:
We are facing an other issue referenced in case #1026

DorianHas, 2020-02-10:
Hello there !

Ok with the new code. Issue seems to be solved. 

Thank you !

closed 2020-02-10 by DorianHas