ControlSystemStudio/phoebus #1026: Building error - MacroHierarchyUnitTest failed

DorianHas, 2019-12-16:
Hello there,

This issue show up 

`
[ERROR] Failures: 
[ERROR]   MacroHierarchyUnitTest.testMacroHierarchy:44 
Expected: "Value from Preferences"
     but: was null
[ERROR]   MacroHierarchyUnitTest.testPropertiesAndEnvironment:97 
Expected: "Value from Preferences"
     but: was null
[INFO] 
[ERROR] Tests run: 69, Failures: 2, Errors: 0, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] app-display-model ................................. FAILURE [1:02.505s]
[INFO] app-display-representation ........................ SKIPPED
[INFO] app-3dViewer ...................................... SKIPPED
[INFO] app-display-representation-javafx ................. SKIPPED
[INFO] app-display-runtime ............................... SKIPPED
[INFO] app-display-editor ................................ SKIPPED
[INFO] app-display-convert-medm .......................... SKIPPED
[INFO] app-display-convert-edm ........................... SKIPPED
[INFO] app-alarm ......................................... SKIPPED
[INFO] app-alarm-model ................................... SKIPPED
[INFO] app-alarm-ui ...................................... SKIPPED
[INFO] app-alarm-logging-ui .............................. SKIPPED
[INFO] app-scan .......................................... SKIPPED
[INFO] app-scan-model .................................... SKIPPED
[INFO] app-scan-client ................................... SKIPPED
[INFO] app-scan-ui ....................................... SKIPPED
[INFO] app-channel ....................................... SKIPPED
[INFO] app-channel-channelfinder ......................... SKIPPED
[INFO] app-channel-utility ............................... SKIPPED
[INFO] app-channel-views ................................. SKIPPED
[INFO] app-perfmon ....................................... SKIPPED
[INFO] app-pace .......................................... SKIPPED
[INFO] app-save-and-restore .............................. SKIPPED
[INFO] app-save-and-restore-model ........................ SKIPPED
[INFO] app-save-and-restore-ui ........................... SKIPPED
[INFO] app-console ....................................... SKIPPED
[INFO] product ........................................... SKIPPED
[INFO] services .......................................... SKIPPED
[INFO] service-alarm-server .............................. SKIPPED
[INFO] service-archive-engine ............................ SKIPPED
[INFO] service-scan-server ............................... SKIPPED
[INFO] service-alarm-logger .............................. SKIPPED
[INFO] service-alarm-config-logger ....................... SKIPPED
[INFO] service-save-and-restore .......................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 1:03.152s
[INFO] Finished at: Mon Dec 16 16:20:52 CET 2019
[INFO] Final Memory: 24M/90M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.20:test (default-test) on project app-display-model: There are test failures.
[ERROR] 
[ERROR] Please refer to /home/hasbroucq/Phoebus/phoebus/app/display/model/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date]-jvmRun[N].dump, [date].dumpstream and [date]-jvmRun[N].dumpstream.
[ERROR] -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.20:test (default-test) on project app-display-model: There are test failures.

Please refer to /home/hasbroucq/Phoebus/phoebus/app/display/model/target/surefire-reports for the individual test results.
Please refer to dump files (if any exist) [date]-jvmRun[N].dump, [date].dumpstream and [date]-jvmRun[N].dumpstream.
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:213)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:153)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:145)
        at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject(LifecycleModuleBuilder.java:84)
        at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject(LifecycleModuleBuilder.java:59)
        at org.apache.maven.lifecycle.internal.LifecycleStarter.singleThreadedBuild(LifecycleStarter.java:183)
        at org.apache.maven.lifecycle.internal.LifecycleStarter.execute(LifecycleStarter.java:161)
        at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:320)
        at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:156)
        at org.apache.maven.cli.MavenCli.execute(MavenCli.java:537)
        at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:196)
        at org.apache.maven.cli.MavenCli.main(MavenCli.java:141)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:290)
        at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:230)
        at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:414)
        at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:357)
Caused by: org.apache.maven.plugin.MojoFailureException: There are test failures.

Please refer to /home/hasbroucq/Phoebus/phoebus/app/display/model/target/surefire-reports for the individual test results.
Please refer to dump files (if any exist) [date]-jvmRun[N].dump, [date].dumpstream and [date]-jvmRun[N].dumpstream.
        at org.apache.maven.plugin.surefire.SurefireHelper.throwException(SurefireHelper.java:204)
        at org.apache.maven.plugin.surefire.SurefireHelper.reportExecution(SurefireHelper.java:98)
        at org.apache.maven.plugin.surefire.SurefirePlugin.handleSummary(SurefirePlugin.java:334)
        at org.apache.maven.plugin.surefire.AbstractSurefireMojo.executeAfterPreconditionsChecked(AbstractSurefireMojo.java:937)
        at org.apache.maven.plugin.surefire.AbstractSurefireMojo.execute(AbstractSurefireMojo.java:785)
        at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo(DefaultBuildPluginManager.java:101)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:209)
        ... 19 more
`

First I though it was my project that was broken so I have fully reinstalled phoebus. This issue is still showing up. I tried on another computer :
- Install project
- Build it
- Start Phoebus (everything is fine)
- Close Phoebus
- Set up eclipse
- Import Maven phoebus project
- Change compliance setting
- Change classpath (replacing app-3d-viewer by app-3dViewer)
- Try to rebuild ==> error showed up here

I tried to delete the .m2 file
Didn't change anything

I think it is related to https://github.com/shroffk/phoebus/issues/82 except we are using CentOS 7 environment.

I assume it is a Maven issue but I can't figure out why.

Regards,
Dorian

shroffk, 2019-12-16:
Hi,
Not really sure why this would be maven issue... looks like the unit tests are failing.

Can you run the test from the IDE or with the -e option to gather more information on why they are failing.

I tried running the tests on my windows machine and there was not issue.

kasemir, 2019-12-16:
>   Install project
>   Build it
>    Start Phoebus (everything is fine)

Did you build that with maven on the command line?

>    Set up eclipse
>    Import Maven phoebus project
I just import as plain Eclipse project, https://github.com/shroffk/phoebus#use-plain-java-configuration, not as maven project, so can't comment on why the import as a maven project would require you to then do this:

>    Change compliance setting
>    Change classpath (replacing app-3d-viewer by app-3dViewer)

.. and fail in the end:

>    Try to rebuild ==> error showed up here

The failure is in the first module that it builds, executing the first test, so I assume something is broken in the Eclipse/maven build.

kasemir, 2019-12-16:
Since you can build with maven on the command line, you can use that product.
If the Eclipse maven import gives you problems, you can either
a) Import as plain eclipse project
b) Use vi, emacs, gedit, notepad, netbeans, inellij, ... instead of Eclipse

shroffk, 2019-12-16:
> Change compliance setting

I checked the parent again and it seems like java source and target are correctly defined to use 11. It might be that eclipse is messing up the .classpath file to require you to explicitly reset the compliance. Normally, in eclipse, I try to disable project specific settings.

> Change classpath (replacing app-3d-viewer by app-3dViewer  

This is something we should fix... @kasemir can we rename/refactor this so that the project folder matches the module name. I am fine either of the two options above. please tell me the one you prefer and I shall update it.

>Try to rebuild ==> error showed up here

I tried importing the display/model and run the tests using m2eclipse. 
Run As --> Maven Install and Tests
How are you trying to rebuild it in eclipse.

kasemir, 2019-12-16:
app-3dViewer is the only project that has an uppercase letter it its name (at least in some of the configs).
So I suggest to change it into app-3d-viewer for all references.

kasemir, 2019-12-16:
I can rename that tomorrow, heading out now.

shroffk, 2019-12-16:
I just saw your last message... oops Sorry I updated it a bit

DorianHas, 2019-12-17:
From what I've tested it seems that it comes from .p2 or .phoebus directories

I deleted these directories and it built well.

kasemir, 2019-12-17:
@kunal, in your renaming commit you missed the top-level build.xml

kasemir, 2019-12-17:
@kunal, since  you already started to rename, I suggest you continue, pull, and I then fix the ant and eclipse part.

shroffk, 2019-12-17:
[image]

I updated the parent build.xml

kasemir, 2019-12-17:
Good, go ahead and pull/merge.

shroffk, 2019-12-17:
This was in the second commit... am I still missing something.
I would like to be more conscientious with not breaking the ant stuff.

kasemir, 2019-12-17:
Go ahead and rename for maven, I can then fix the ant and eclipse build.

shroffk, 2019-12-19:
While as I said before I can't recreate the macro failures...
I do have a few failures with PathTests when I run them in eclipse. 

This is on a windows machine.
```
[INFO] Running org.csstudio.display.builder.PathTest
[ERROR] Tests run: 4, Failures: 2, Errors: 0, Skipped: 0, Time elapsed: 0 s <<< FAILURE! - in org.csstudio.display.builder.PathTest
[ERROR] testRelative(org.csstudio.display.builder.PathTest)  Time elapsed: 0 s  <<< FAILURE!
java.lang.AssertionError: 

Expected: "other.bob"
     but: was "/one/of/my/directories/other.bob"
	at org.csstudio.display.builder.PathTest.testRelative(PathTest.java:77)

[ERROR] testCombine(org.csstudio.display.builder.PathTest)  Time elapsed: 0 s  <<< FAILURE!
java.lang.AssertionError: 

Expected: "/home/beamline/main.bob"
     but: was "https://webopi.sns.gov/webopi/opi//home/beamline/main.bob"
	at org.csstudio.display.builder.PathTest.testCombine(PathTest.java:67)

[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Failures: 
[ERROR]   PathTest.testCombine:67 
Expected: "/home/beamline/main.bob"
     but: was "https://webopi.sns.gov/webopi/opi//home/beamline/main.bob"
[ERROR]   PathTest.testRelative:77 
Expected: "other.bob"
     but: was "/one/of/my/directories/other.bob"
[INFO] 
[ERROR] Tests run: 69, Failures: 2, Errors: 0, Skipped: 0
```

shroffk, 2019-12-19:
The above errors lead me to the ModelResourceUtil and I was wondering if there is a reason why Path from java.nio package and URI are not used.
Operations like normalize, relativize, get path segments, get parents... are all supported in some for or the other. 
I am sure they will not fit perfectly but is there a reason to reimplement these from scratch instead of building around the standard java packages?

kasemir, 2019-12-19:
Feel free to re-implement based on java.nio.Path, but note that it has to handle both http://.. and file paths, so java.nio.Path might not be a plug-in replacement.
The requirement is basically that it should "work" with those paths which people have entered over the last 10 years in their display files for embedded and related displays. I think the unit test covers the key scenarios of resolving absolute and relative paths relationships. The test might indeed not pass on Windows, though I'm not aware of issues with resolving actual screens on windows, so not sure why the test fails.

shroffk, 2019-12-19:
> so java.nio.Path might not be a plug-in replacement.

Yes, I concluded that with my overview

>  though I'm not aware of issues with resolving actual screens on windows

Me neither

kasemir, 2019-12-19:
@DorianHas , back to the original issue:

You can build with maven on the command line, can't you just use that to move on with #1007?
If you do want to work on the sources, seems like everybody else is now moving to IntelliJ, so try that.

Or try Eclipse without maven import, just follow https://github.com/shroffk/phoebus/blob/master/README.md#use-plain-java-configuration

We should not turn this into a ticket for debugging the Eclipse maven support.

DorianHas, 2020-02-10:
Hello there !

We moved to Intellij. It solved the issue.

Thank you

closed 2020-02-10 by DorianHas

shroffk, 2020-02-12:
#1136