ControlSystemStudio/phoebus #1032: PVA structure time stamps

shroffk, 2019-12-17:
In pvaccess, when references a part of a larger structure there is a possibility that the sub structure will not have a timestamp. 

eg.
Complete Structure:

[image]

Substructure being used in cs-studio/phoebus:

[image]

When the substructure is used in Display builder...despite the fact that the value is changing, since the time stamp remains the same the widget does not update. (I am not sure if the event is not being delivered to the widget or being discarded due to the same timestamp)

Ideas:
- core-pv/pva should somehow extract the timestamp from the parent structure
- display builder, like probe should respond to value changes even when the time stamp is the same.
- we have a flag with just adds the current time stamp

kasemir, 2019-12-17:
I don't think it's a timestamp issue. It's a matter of getting or not getting events. 
Increase the log level of org.epics.pva to FINE and check if you get "Received monitor #.. update for <channel name>" messages in one case but not the other.

kasemir, 2019-12-17:
What are you using as a pva server? Is that something I can reproduce locally?

shroffk, 2019-12-18:
Ok, 
Ignore the above report. After some digging the problem is not to associated with the timestamp.

The problem lies in the value comparison in the PropertyChangeHandler

```
       // Any change at all?
       if (new_value != null  &&  old_value != null  &&  new_value.equals(old_value))
           return;
```

In my case, the value is a IVDoubleArray... where the equality is checked by checking the underlying raw data points to the same array instead of looping through the array values.

```
    @Override
    public boolean equals(Object obj) {
        if (obj == this)
            return true;

        if (obj instanceof ArrayDouble) {
            ArrayDouble other = (ArrayDouble) obj;
            
            if ((array == other.array) && startIndex == other.startIndex && size == other.size)
                return true;
        }

        return super.equals(obj);
    }

```

Despite having different array values the arrays in the old and new are the same. 

[image]

shroffk, 2019-12-18:
The server is available here:
https://github.com/shroffk/scope

kasemir, 2019-12-18:
Does the array data indeed not change, so that part would be correct?

Maybe the issue here is that IVDoubleArray.equals returns True when it finds that the array data matches, instead of also checking if the alarm and time stamp match.

kasemir, 2019-12-18:
If the data does change, you should see the debugger step into `PVADoubleArray.decode()`.
It currently keeps the array and only updates the elements as long as the array size stays the same.
I guess that's wrong. Should get a new array (as done in `setValue`) because who knows who is using that array and assumes it's the same as long as the array itself is the same.

kasemir, 2019-12-18:
Maybe try changing this
```
    public void decode(final PVATypeRegistry types, final ByteBuffer buffer) throws Exception
    {
        final int size = PVASize.decodeSize(buffer);
        double[] new_value = value;
        if (new_value == null  ||  new_value.length != size)
            new_value = new double[size];
        for (int i=0; i<size; ++i)
            new_value[i] = buffer.getDouble();
        value = new_value;
    }
```

into this and see if that fixes it:

```
    public void decode(final PVATypeRegistry types, final ByteBuffer buffer) throws Exception
    {
        final int size = PVASize.decodeSize(buffer);
        final double[] new_value = new double[size];
        for (int i=0; i<size; ++i)
            new_value[i] = buffer.getDouble();
        value = new_value;
    }
```

shroffk, 2019-12-18:
>I guess that's wrong. Should get a new array (as done in setValue) because who knows who is using that array and assumes it's the same as long as the array itself is the same.

Yes this is my opinion.

I have verified that the data does indeed change... not in the decode but by putting a breakpoint and looking into the data in the ArrayDouble

shroffk, 2019-12-18:
I will try the above

shroffk, 2019-12-18:
>> assumes it's the same as long as the array itself is the same. 

Was this sarcastic :)
Anyway, as expected the creation of a new array addresses the above problem.

kasemir, 2019-12-18:
Why would I be sarcastic?  An array VType consists not only of the array, but also time stamp, alarm, display info. IVDoubleArray.equals assumes equality just based on the array, not checking time stamp etc.
Anyway, thanks for finding and fixing the PVA client issue.

shroffk, 2019-12-18:
Oh no no...It does check all of those. Which is why when I played around with new timestamp the updates started coming.

That is part of the VNumberArray.
```
	if (obj instanceof VNumberArray) {
            VNumberArray other = (VNumberArray) obj;
        
            return getClass().equals(other.getClass()) &&
                    getData().equals(other.getData()) &&
                    getAlarm().equals(other.getAlarm()) &&
                    getTime().equals(other.getTime()) &&
                    getDisplay().equals(other.getDisplay());
        }
        
```

The data part is where we were running into trouble...

```
getData().equals(other.getData()) &&
```
In the above case all the rest were the same and only the array data was different.

closed 2019-12-18 by shroffk

kasemir, 2019-12-18:
OK, so we're overall good, thanks!