ControlSystemStudio/phoebus #1033: Restore preference to default value does not work.

georgweiss, 2019-12-18:
Do not know if this is intentional or not, but a preference setting that is overridden in an ini file will retain its value even if the setting in question is removed from the ini file. Guess this is because it is read from prefs.xml. If I remove .phoebus the default value is correctly set.

kasemir, 2019-12-18:
See https://phoebus-doc.readthedocs.io/en/latest/preferences.html:
"Settings loaded via either the -settings .. command line option or the Import Preferences button are stored in the user location (see Locations). They remain effective until different settings are loaded or the user location is deleted."

georgweiss, 2019-12-18:
Not sure I understand. If I remove a setting in the ini file the value is not restored to the default even if I reload the edited ini file by means of the Import Settings button.

kasemir, 2019-12-18:
Your settings are kept in the user location (default: $HOME/.phoebus).
If you load a file which sets "some.package/setting=X", that's stored in the settings.
If you then load a file which doesn't contain anything for "some.package/setting", it remains at X.

georgweiss, 2019-12-18:
Exactly my point: if I for any reason wish to restore to the default value I need to wipe the .phoebus directory or find the prefs.xml file containing it.
What is the purpose of the prefs.xml files anyway? After all, the contain the same information as the default values from the preferences files, plus overridden values from ini file. There is no editor manipulating the prefs.xml files.

kasemir, 2019-12-18:
Going back to the original design of Phoebus, i.e. trying to move on from Eclipse, we decided to use java preferences instead of inventing something new.
https://docs.google.com/document/d/11W52PRlsRjpIvP81HxUxxR9g180DHDByCohYQ9TQv7U

Then we found that while the Linux JRE saves those preferences in files (XML) where we can control the location, the Windows JRE would use the system registry. So we register a copy of the Linux FileSystemPreferences as a PreferencesFactory to get the same behavior on all platforms.

Finally, we liked to have some way to document the available settings and their default values. So we added the PreferencesReader. It supports a "property file" format since that's very similar to what we had used with Eclipse, it allows allows for comments that describe the meaning and possible values for each settings.

Still, the basic mechanism remains unchanged:
Loading `-settings some_file.ini` loads the settings in that file into the java preferences, and all the rest then reads from the java prefs API.
If you later remove settings from that file, you get the last persisted value for those settings.

If you want to change that behavior, I guess you can't use the java preferences. At minimum you'd have to remove its persistence, so each time it would load the defaults, then apply the `-settings`.  .. except the time line is currently different: The phoebus launcher first applies the `-settings`, and then each application that's started might load its `*_preferences.properties` for those settings that are not already in the java settings. So each application in its `start()` would need to populate the settings with their defaults, then you can load the `-settings`. This means each application would have to access their settings file at startup, even when the app is never used. If you do that right, you might be able to get to the Eclipse startup time.

kasemir, 2019-12-18:
Maybe explain which actual setting is causing you problems?
In our operational practice, we see two types of settings:
a) The default is OK, no need to change at all.
b) An operational setup always needs to set this, because the default is "localhost".

So in practice, the settings file contains things that we need to set.
We might change some value, but not remove a value. Going back to 'localhost' won't get us PVs, archived data, scan server access, ...

georgweiss, 2019-12-18:
To give you an example: org.phoebus.pv/default. I do some tests to verify that setting it to pva works in my code. When I'm done I'd like to switch back to ca knowing that it's the default. I can change it in my ini file, but I was expecting it to work by removing the line as I know ca is the default.

kasemir, 2019-12-18:
Right. You'd have to once load the file with `org.phoebus.pv/default=ca`, and then you could remove that line.

Changing the network protocol used at your site is not a thing that an end user would frequently do.
On the other hand, if you have one beamline using ca, and another beamline using pva, then that would be a good reason to always have that setting in the beamline-specific ini file.

georgweiss, 2019-12-18:
Agree my use case is rare, but this applies to all settings. I would assume operators should not touch the ini file very frequently, but there are other type of users that need more control.
I can live with the current implementation. I will propose an update to the documentation to clarify how to restore default values.

closed 2019-12-19 by georgweiss