ControlSystemStudio/phoebus #1040: Widget version numbers

krisztianloki, 2019-12-20:
As far as I understand widget version numbers are only used to achieve backwards compatibility (i.e. loading .opi files). What about forwards compatibility; for example refusing to run/load the display if any of the widgets' major version number is higher than what we support? This could be further extended; minor version incompatibility would mean that the display can be run but cannot be edited/saved.
So an increase in the major version number would mean an incompatible change in runtime behavior, an increase in the minor version number would mean a runtime compatible change in the on-disk format (for example a new property that is not used / not critical for representation/runtime).
What do you guys think?

kasemir, 2019-12-20:
The code for each widget can decide how it handles the different versions.
For now, some widgets use the version information to be backwards compatible.
You're correct that the code for specific widgets could also refuse to read newer display files.
You could for example email all displays in question to a local display adjustment expert.
But what is your actual use case? Do you have a problem running displays in a production setup?

krisztianloki, 2019-12-20:
I don't have a problem running displays but I can think of a couple of scenarios where we would have problems editing them.
Take for example this issue from Claudio: [329](https://github.com/kasemir/org.csstudio.display.builder/issues/329). This is something that would not change the runtime behavior or the representation; however, editing and saving a display in an unsupported Phoebus version (one that does not recognize this new property) would lead to losing this information. Also, I would like to have another feature similar to the "Notes property"; listing the macros that are used in a display with a short description of what they represent. So, what I am saying is that if there is no way to prevent modifying a display with an "outdated" Phoebus version, there is no way to add new features that extend the current xml format.