ControlSystemStudio/phoebus #1042: Properties and values committed only after hitting ENTER

georgweiss, 2019-12-30:
In most cases the user must press the ENTER key to commit edited values when updating properties and other entities. This is annoying, in particular as this is not always needed (e.g. color picker). Further, this is contrary to the behavior in most commercial SW.

Is there any reason to not update properties and others on lost focus?

kasemir, 2019-12-30:
They behave the same way as text input fields: User needs to confirm via enter. Pressing ESC or leaving the focus is taken as change-of-mind,restoring the original value.

georgweiss, 2020-01-02:
For a text input field in an OPI that makes sense, but I fail to see the purpose of this behavior for input fields of the editor. Our users are complaining about this, and I fully support them. I cannot think of any SW where I have seen the same behavior in a comparable context.

pavel-ch, 2020-01-02:
I would also ask for a change in editor, this behavior
> leaving the focus is taken as change-of-mind

confused me several times and is non-standard comparing to other software I know using a kind of editable property grid. ESC is behaving correctly here.
In OPIs, on the contrary, this general rule
> User needs to confirm via enter

is a desired and expected behavior and no change is needed.

pavel-ch, 2020-01-07:
Can we close the issue @georgweiss ?

georgweiss, 2020-01-08:
Not quite. Even though your fix works fine for widget properties, I'd like to see the same behavior in for instance the rule editor. Currently, if a new rule is added, a modified name must be committed by hitting Enter. Same with the other entities in the rule editor. The script editor and macro editor show the same behavior that I would like to change.
If you do not have the bandwidth to fix those editors, I can do it.

pavel-ch, 2020-01-08:
Yes you are right, please continue by yourself, thanks.
Be aware of #893 (see my question yesterday).

pavel-ch, 2020-02-13:
Still missing at least Macros dialog and Points dialog.

georgweiss, 2020-02-13:
I intentionally left out the Macros dialog since I do not think a change adds much value.

pavel-ch, 2020-02-13:
Do not mind if I do it? I do not like the current state ... - at least it is an irregularity now.

kasemir, 2020-02-13:
It was pretty consistently using 'ENTER'. Want to change that, fine, but then I agree that it should again be consistent, so thank you Pavel for picking this up and completing it.

georgweiss, 2020-02-13:
No I do not mind.

closed 2020-03-16 by pavel-ch