ControlSystemStudio/phoebus #1043: Text color conversion in Display Builder

tanviash, 2020-01-02:
For BOY, the text color inside the boxes automatically changed from black to white and vice versa depending on the textbox color. However, this does not seem to happen in Phoebus Display Builder automatically. See the image below.

[image]

kasemir, 2020-01-08:
Yes, the text color of Display Builder buttons is simply the "foreground_color" property, it's not magically changing based on the "background_color".
Can you attach the above *.opi file? The *.opi import might be able to set the initial "foreground_color" based on the "background_color" to black/white.

tanviash, 2020-01-08:
@kasemir Here is the opi file 
[multi-color-leds-opi.txt](https://github.com/shroffk/phoebus/files/4037143/multi-color-leds-opi.txt)

Two issues with multi-color LEDS:
1)	Text color black/white depending on background color
2)	No 3d effect, which give the background color gradient (less important, but nice to have)

Also progress bar widget does not show.

kasemir, 2020-01-09:
On the progress bar, open the *.opi in the BOY editor and check the size.
It's actually about twice as high as shown, the upper half of the widget is below the "Duty Factor ..." row:

[image]

 The Display Builder progress bar uses the actual widget size, and completely fills it.
This is what it looks like when the PV value is around 80:

[image]

And yes, the Display Builder progress bar is for now just a bar, no scale.

kasemir, 2020-01-09:
Better comparison of BOY and Display builder side by site, to show how the sizes compare:

[image]

I'll look into shrinking the BOY progress bar height when importing it, but will have to guess how high BOY would actually render it.

dylan171, 2020-01-09:
For the MutliState LED in BOY, the label color (either black or white) is determined by this [formula](https://github.com/ControlSystemStudio/cs-studio/blob/dbcf552ff2fb442e1d97c89a4835ce245468cc59/applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets/src/org/csstudio/swt/widgets/figures/LEDFigure.java#L385) to ensure that it always has good contrast with the background color.

kasemir, 2020-01-09:
Right, and for the display builder LEDs you can configure a color. I don't want to simply ignore that color, but only replace it with black resp. white when the color of the text and the color of the led are too close in their brightness. So it'll automagically change color at some point to black/white, but should generally use the configured color.

dylan171, 2020-01-09:
Great, that would be even better!

kasemir, 2020-01-14:
Close?

closed 2020-01-14 by tanviash