ControlSystemStudio/phoebus #1044: Confirm Message for Action button

tanviash, 2020-01-02:
Action button in Phoebus does not ask the confirm message:
[image]

kasemir, 2020-01-08:
BOY supported a confirmation message on individual actions.
The Display Builder supports similar confirmation messages, but on the Action Button.

I can update the compatibility code such that when opening an *.opi action button where the action has a confirmation message, that's transferred into the button settings, so in the end pressing the button will again prompt for the same confirmation message.
Can you attach a simple example *.opi with just one button and the above action?

tanviash, 2020-01-08:
@kasemir Here is the file that has the Action Button widget. Search for "Reset the crate".
[confirm-message-opi.txt](https://github.com/shroffk/phoebus/files/4037137/confirm-message-opi.txt)

kasemir, 2020-01-09:
This should now work, try with the latest

kasemir, 2020-01-14:
Close?

closed 2020-01-14 by tanviash