ControlSystemStudio/phoebus #1045: Ant build broken for alarm logger?

MaticPogacnik, 2020-01-05:
I was using alarm logger maven build for few months now, and everything worked ok, with exception of few warrnings:

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.springframework.cglib.core.ReflectUtils$1 (file:/home/mpogacnik/Phoebus/release4.6.0/phoebus-parent-4.6.0/services/alarm-logger/target/lib/spring-core-5.0.9.RELEASE.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of org.springframework.cglib.core.ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
2020-01-05 11:29:29.146  WARN 998 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_port' was supplied but isn't a known config.
2020-01-05 11:29:29.142  WARN 998 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_port' was supplied but isn't a known config.
2020-01-05 11:29:29.152  WARN 998 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2020-01-05 11:29:29.152  WARN 998 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_host' was supplied but isn't a known config.
2020-01-05 11:29:29.152  WARN 998 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2020-01-05 11:29:29.153  WARN 998 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2020-01-05 11:29:29.153  WARN 998 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2020-01-05 11:29:29.154  WARN 998 --- [pool-1-thread-1] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2020-01-05 11:29:29.153  WARN 998 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_host' was supplied but isn't a known config.
2020-01-05 11:29:29.155  WARN 998 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2020-01-05 11:29:29.156  WARN 998 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2020-01-05 11:29:29.156  WARN 998 --- [pool-1-thread-2] o.a.k.clients.admin.AdminClientConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2020-01-05 11:29:29.272  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2020-01-05 11:29:29.272  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2020-01-05 11:29:29.273  WARN 998 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2020-01-05 11:29:29.273  WARN 998 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2020-01-05 11:29:29.273  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2020-01-05 11:29:29.273  WARN 998 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2020-01-05 11:29:29.273  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2020-01-05 11:29:29.274  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2020-01-05 11:29:29.274  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2020-01-05 11:29:29.274  WARN 998 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2020-01-05 11:29:29.275  WARN 998 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2020-01-05 11:29:29.276  WARN 998 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2020-01-05 11:29:29.363  WARN 998 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2020-01-05 11:29:29.363  WARN 998 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2020-01-05 11:29:29.363  WARN 998 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2020-01-05 11:29:29.363  WARN 998 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2020-01-05 11:29:29.363  WARN 998 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2020-01-05 11:29:29.363  WARN 998 --- [pool-1-thread-1] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2020-01-05 11:29:29.368  WARN 998 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2020-01-05 11:29:29.368  WARN 998 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2020-01-05 11:29:29.368  WARN 998 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2020-01-05 11:29:29.368  WARN 998 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2020-01-05 11:29:29.369  WARN 998 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2020-01-05 11:29:29.369  WARN 998 --- [pool-1-thread-2] o.a.k.clients.producer.ProducerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2020-01-05 11:29:29.430  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
2020-01-05 11:29:29.430  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_value' was supplied but isn't a known config.
2020-01-05 11:29:29.431  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_host' was supplied but isn't a known config.
2020-01-05 11:29:29.431  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'date_span_units' was supplied but isn't a known config.
2020-01-05 11:29:29.433  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_port' was supplied but isn't a known config.
2020-01-05 11:29:29.433  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'admin.retries' was supplied but isn't a known config.
2020-01-05 11:29:29.433  WARN 998 --- [pool-1-thread-1] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'alarm_topics' was supplied but isn't a known config.
2020-01-05 11:29:29.436  WARN 998 --- [pool-1-thread-2] o.a.k.clients.consumer.ConsumerConfig    : The configuration 'es_max_size' was supplied but isn't a known config.
[... cut ...]
Caused by: java.lang.ClassNotFoundException: org.elasticsearch.script.mustache.SearchTemplateRequest
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
        ... 59 more
```
I am getting error that logger implementation is missing and an NoClassDefFoundError exception.
Any idea why this is happening? Thanks!

MaticPogacnik, 2020-01-05:
Maven build still seams to work. It fails after I run:

```
ant clean dist
```

shroffk, 2020-01-06:
> Maven build still seams to work. It fails after I run:
> 
> ```
> ant clean dist
> ```

What is the point of this step...

If you are ok with using maven... all you need to do to build everything is 

```
wget https://github.com/shroffk/phoebus/archive/parent-4.6.0.zip
unzip parent-4.6.0.zip
cd phoebus-parent-4.6.0
mvn clean install -DskipTests
```

you don't need to run both the maven and the ant commands...

kasemir, 2020-01-08:
Kunal is correct, you don't have to build with ant. You can build with maven if that's what you prefer.
The idea of the ant built is that it's 100% local with known dependencies, and the 'dist' step creates zip files with start script, online help etc. as appropriate.

Our goal, I thought, would be to support both, see https://github.com/shroffk/phoebus/tree/master/dependencies, and to know what our dependencies are.

Your specific error about the missing `*.mustache.*` class is caused by a new dependency that was not listed in the ant built file. I'll soon have a PR that updates the build.xml for the alarm logger to list all the new jar files.
But right now our target platform is incomplete. When you build the alarm-logger with maven, it does that with these:
```
commons-codec-1.11.jar
connect-api-1.0.2.jar
connect-json-1.0.2.jar
httpasyncclient-4.1.4.jar
httpclient-4.5.6.jar
httpcore-4.4.10.jar
httpcore-nio-4.4.10.jar
jackson-annotations-2.9.0.jar
jackson-core-2.9.6.jar
jackson-databind-2.9.6.jar
jackson-dataformat-cbor-2.9.6.jar
jackson-dataformat-smile-2.9.6.jar
jackson-dataformat-yaml-2.9.6.jar
jackson-datatype-jsr310-2.9.6.jar
joda-time-2.9.9.jar
log4j-api-2.10.0.jar
```

Our target platform includes these:
```
commons-codec-1.10.jar
connect-api-2.0.0.jar
connect-json-2.0.0.jar
httpasyncclient-4.1.2.jar
httpclient-4.5.2.jar
httpcore-4.4.5.jar
httpcore-nio-4.4.5.jar
jackson-annotations-2.10.1.jar
jackson-core-2.10.1.jar
jackson-databind-2.10.1.jar
jackson-dataformat-cbor-2.8.10.jar
jackson-dataformat-smile-2.8.10.jar
jackson-dataformat-yaml-2.8.10.jar
jackson-datatype-jsr310-2.10.1.jar
joda-time-2.10.jar
log4j-api-2.11.1.jar
```
Some versions are newer, others are older.

MaticPogacnik, 2020-01-08:
Hey,

Thank you for reponses.
Yes I know I can use Maven, but I prefer ant just for the stated reason - It creates zip file which I can nicely distibutate (in case of a alarm logger I use the zip as the input to the Docker contaier where it runs).

Is it possible to create zip with Maven? (Until now I packed it manualy, which is a nasty way of doing it I think.)

Thanks!

kasemir, 2020-01-08:
Kunal, can you consolidate the alarm logger dependencies with the target platform?
Might be a matter of picking consistent versions for spring.boot, elastic, kafka.

shroffk, 2020-01-22:
> Is it possible to create zip with Maven? (Until now I packed it manualy, which is a nasty way of doing it I think.)

yes, I will add this functionality

> Kunal, can you consolidate the alarm logger dependencies with the target platform?
Might be a matter of picking consistent versions for spring.boot, elastic, kafka.

Ok.

shroffk, 2020-01-22:
I will coordinate with tanvi & george to get the dependencies in order.

shroffk, 2020-02-03:
#1100 this PR includes profiles and plugins which creates tar and zip s for alarm-logger & alarm-config-logger services. It also provides a profile to package the common phoebus product.

shroffk, 2020-02-26:
@MaticPogacnik can we close this?
The PR #1100 should have addressed your request

MaticPogacnik, 2020-02-26:
Yes, this can be closed now maven packeging looks great! Thanks.

closed 2020-02-26 by MaticPogacnik