ControlSystemStudio/phoebus #1065: PostgresSQL schema for archive-engine?

mhedges, 2020-01-14:
Are there any plans to include an example schema for PostgresSQL database for use with the `archive-engine`?  The existing `MySQL.db` is great, but it would also be nice to have similar for PostgresSQL.

kasemir, 2020-01-14:
There's no change to the SQL, so you can try the schemas that were published in the Eclipse-based version, https://github.com/ControlSystemStudio/cs-studio/tree/master/applications/archive/archive-plugins/org.csstudio.archive.rdb/dbd

I've included the MySQL.dbd in https://github.com/shroffk/phoebus/tree/master/services/archive-engine/dbd, because I've actually used that one when testing the Phoebus version.
I'm not running Postgres, so I didn't simply copy those over. Also note that with Postgres, there are at least two different ways to set it up. You can have a plain 'samples' table, which is easy to get started, but unlikely sufficient to support a large, long-running setup. For that you want to partition the samples, for example periodically like each month, and maybe also by channel ID.

Can you try the basic postgres_schema.txt, and if that works OK, we'll copy it over to the new archive-engine sources? Also feel free to try and maybe update the postgres partitioning example, but I can't help you with that other than include the end result as an example when you verified that it's useful.

mhedges, 2020-01-15:
Ok, thank you very much.  I will start here and report back with what I find out.

mhedges, 2020-01-21:
I think I was able to get this to work.  The only important thing when using the `postgres_schema.txt` is that I had to run lines 38-61 at the end, or else there's an error about access permissions.

I think this satisfies my needs. I will report back if I find any other issues.

kasemir, 2020-01-22:
Thanks for checking it, I've imported the postgres_schema.txt

mhedges, 2020-01-23:
Thanks!

For posterity (in case anyone ends up here also wanting to use this), I should probably mention that I also needed to modify https://github.com/shroffk/phoebus/tree/master/services/archive-engine/src/main/java/org/csstudio/archive/Engine.java
and https://github.com/shroffk/phoebus/tree/master/services/archive-engine/src/main/java/org/csstudio/archive/Engine.java to use PostgreSQL and then recompile.

kasemir, 2020-01-23:
Can you send diffs or details on what you had to change?

mhedges, 2020-01-23:
Turns out all I had to change was `services/archive-engine/src/main/resources/archive_preferences.properties`:

```
diff --git a/services/archive-engine/src/main/resources/archive_preferences.properties b/services/archive-engine/src/main/resources/archive_preferences.properties
index 02fa6c2c3..74769c43f 100644
--- a/services/archive-engine/src/main/resources/archive_preferences.properties
+++ b/services/archive-engine/src/main/resources/archive_preferences.properties
@@ -10,6 +10,9 @@
 # MySQL example
 url=jdbc:mysql://localhost/archive?rewriteBatchedStatements=true

+# PostgreSQL example
+url=jdbc:postgresql://localhost/archive
+
 # RDB user and password
 # Some applications also provide command-line option to override.
 user=archive
```

kasemir, 2020-01-24:
Sure, you'll have to configure the URL. In a production setup it'll likely be some host other than localhost.
I'll add that example URL, #1078 

Note that you shouldn't have to edit the build-in archive_preferences and recompile.
Instead, you create a file my_settings.ini:
```
# My archive URL
org.csstudio.archive/url=jdbc:postgresql://my_postgres_host.site.org/archive

# Likely also need to configure this:
org.phoebus.pv.ca/addr_list=my_gateway_host.site.org some.other.ip.address 160.91.228.17
```
and then start the archive engine with `-settings /path/to/my_settings.ini`

mhedges, 2020-01-27:
Ok, great!  That's even easier.  Thanks for the tip

closed 2020-03-02 by kasemir